syssi/esphome-yeelight-ceiling-light #9: Yeelight Ceiling Light Bluetooth Remote

jaddel, 2021-03-17:
Did you already made any attempts to get the Bluetooth remote working?

syssi, 2021-03-17:
No. This will be hard. The first step is to get bluetooth running at esphome. My dream is a `esp32_ble_tracker` per yeelight device / ceiling lamp. It looks like I've to replace the ESP32 with a dual core version first. As soon I'm able to enable the btle stack at the esp32 we can try to communicate with btle devices. Are you aware of [this](https://github.com/nccchirag/yeelight-ble-rotary-dimmer/issues/1) project?

jaddel, 2021-05-14:
> h a dual core version first. As soon I'm able to enable the btle stack at the esp32 we can try to communicate with btle devices. Are you

i skipped through it but I'm not familiar with the matter. I looks like Tasmota is already supporting the remote? #yee-rc
https://tasmota.github.io/docs/Bluetooth/#ble-sensors-on-esp32-using-built-in-bluetooth

syssi, 2021-05-14:
Good catch! I've replaced the ESP32 of one of my lamps by a dual core version already and will give it a try.

Ernst79, 2021-05-14:
The remote is not encrypted. I recently added it to BLE monitor (it's added and working in 2.0.1-beta, not in master yet). https://github.com/custom-components/ble_monitor/issues/289

Note that in that topic, we discuss the Yeelight remote (which is not encrypted) and the rotating dimmer. the last is not clear, it seems not to follow the MiBeacon protocol, at least it has has a strange format.

jaddel, 2021-05-14:
awesome, thx. Do you know if this also natively possible using esphome?

Ernst79, 2021-05-14:
The info was meant for @syssi to implement it in ESPhome. So it’s not yet in ESPhome, but with the info, it should be easy to add for him, I assume.

syssi, 2021-05-15:
I've prepared a esphome PR: https://github.com/esphome/esphome/pull/1783

jaddel, 2021-05-18:
Could you give me a hint how i can mange to get your custom integration for the BLE remote to work until your commit has been added to the esphome master?

syssi, 2021-05-18:
The `custom_components` folder is used automatically if you keep the directory structure:

```
custom_components/
your-ceiling-lamp.yaml
```

Just add the relevant parts of the `yeerc_ylyk01yl.yaml` to `your-ceiling-lamp.yaml`: 

```
esp32_ble_tracker:

xiaomi_ylyk01yl:
  mac_address: "FF:FF:FF:FF:FF:FF"
  last_button_pressed:
    name: "last button pressed"
  on_button_on:
    then:
      - logger.log: "Button on pressed"
  on_button_off:
    then:
      - logger.log: "Button off pressed"
  on_button_sun:
    then:
      - logger.log: "Button sun pressed"
  on_button_m:
    then:
      - logger.log: "Button m pressed"
  on_button_plus:
    then:
      - logger.log: "Button plus pressed"
  on_button_minus:
    then:
      - logger.log: "Button minus pressed"
```

You've to update the `mac_address` to the MAC address of your remote control. If you have trouble to find the MAC address I can provide some instructions for a linux host.

Please make sure you apply the above configuration to a ceiling lamp equipped with a **ESP32 dual core**.

jaddel, 2021-05-18:
awesome, it's working!

thank you.

syssi, 2021-05-18:
@jaddel If you ceiling lamp misses some key presses of the remote control you could play around with these settings:

```
esp32_ble_tracker
  scan_parameters:
    interval: 320ms
    window: 100ms
```

Please be aware: If the `esp32_ble_tracker` blocks the radio of the ESP32 too heavily your WiFi connection fails. You could use a separat ESP32-Nodemcu to play around.

jaddel, 2021-05-18:
thx, I'll give it a try. 

For the YLXD05YL where i replaced the Mediatek microcontroller by an ESP32-Nodemcu the remote from my YLXD41YL is working fine at first glance. The original remote is getting recognized by the ble tracker but is not working with your integration. 

Is there something you can recommend trying?

This is the log output for the YLXD05YL Remote: 

> [09:43:17][D][esp32_ble_tracker:544]: Found device F8:24:41:DC:61:C3 RSSI=-79
> [09:43:17][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
> [09:43:19][D][adc:058]: 'Office Ceiling adc1': Got voltage=0.00V
> [09:43:19][D][sensor:099]: 'Office Ceiling adc1': Sending state 0.00000 V with 2 decimals of accuracy
> [09:43:20][D][esp32_ble_tracker:544]: Found device 73:6D:9C:BF:C4:48 RSSI=-94
> [09:43:20][D][esp32_ble_tracker:565]:   Address Type: RANDOM
> [09:43:20][D][esp32_ble_tracker:569]:   TX Power: 2
> [09:43:38][D][esp32_ble_tracker:544]: Found device 60:FB:74:7E:71:A3 RSSI=-72
> [09:43:38][D][esp32_ble_tracker:565]:   Address Type: RANDOM

syssi, 2021-05-18:
Could you provide some details about the second remote control? How does it look like?

jaddel, 2021-05-18:
From the outside 100% the same.

syssi, 2021-05-18:
Do you own a linux host incl. bluetooth LE connectivity? I could provide some instruction later the day to dump the BLE advertisements of the second remote control for comparison.

jaddel, 2021-05-18:
I'll prepare something.

jaddel, 2021-05-18:
would a macbook mid 2015 be sufficient?

syssi, 2021-05-18:
I think so. Do you have an android smartphone around? I assume downloading some app to store bluetooth advertisements could be easier.

syssi, 2021-05-18:
Try to use this app:

https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan

Press "SCAN" und push some buttons of your remote control. If the "yeerc" pops up at the result page please open the entry and provide a screenshot. I need the payload of the section "SCAN RECORD".

[image]

jaddel, 2021-05-18:
working on it.

do you know if there is a way to trigger some kind of save boot without any integrations loaded? Since i played with the scan_paramters the lamp still goes online but connection is to bad for ota uploads. :/

syssi, 2021-05-18:
Do you have access to the reset button of the esp32? If you reset the device 10 times in a row esphome goes into the safe mode (hopefully).

jaddel, 2021-05-18:
not really, then I'll open the light again. For the next time maybe the reset switch component can do the same thing? 

[image]
[image]

rezmus, 2021-05-20:
remote can be also legacy encrypted (like the one from app).

jaddel, 2021-06-15:
Any update on this?