syssi/esphome-yeelight-ceiling-light #10: Request YLXD05YL Support

splitatom44, 2021-04-03:
Is it possible to assist with flashing the YLXD05YL and YLXD01YL models of the Yeelights?
Photos attached of mainboards.
Thankyou

[image]
[image]
[image]
[image]
[image]
[image]
[image]

syssi, 2021-04-03:
Bad news: The heart of the white daughterboard is a Mediathek microcontroller. Esphome has ESP8266 and ESP32 support only. Your device aren't Esphome compatible. :-(

splitatom44, 2021-04-04:
Ah ok, thought that might be the case when I pulled it apart. Thanks anyways

syssi, 2021-04-04:
Do you like to replace the white daughterboard with an esp32? The task isn't easy but possible.

splitatom44, 2021-04-04:
Interesting, I'd assume a fairly good understanding of the schematics of the board would be needed no?

syssi, 2021-04-04:
Yes. I try to outline the steps to get a better undestanding what needs to be done:

1. Use the pads at J2 to power the daughter board. Attach 3.3V + GND to a power supply providing 3.3V. Don't attach the LEDs! **Don't** power the device with 230V. Attach RXD, TXD and GND to a USB-to-TTL serial adapter.
2. Open a serial console. You should see some output of the stock firmware.
3. Try to control the device by the android app. Do you see any output at the serial console?
4. Remove the power. Try to understand / identify which pads of the white daugther board are connected to the led driver. Trace the lines / use a continuity tester. 

We are ready if you've identified the following pins:
- START
- 3.3V
- GND
- RXD
- TXD
- C (Cold white), PWM signal
- W (Warm white), PWM signal
- Night light (if supported), PWM signal
- Standby / Power supply (if supported)

5. As soon we know the location of the pads of the different functions we will try to compare the pin layout of an esp32 wroom. I hope the esp32 can be attached/soldered 1:1 to the board. If not: The relevant pins needs to be wired to the corresponding pin (gnd, 3.3v, txd, rxd) of the esp32.

6. As soon we know/assume the location/pads of C, W, N and Standby we can verify our assumption by driving the stock firmware via the app. F.e. turn on the light and drive cold white to 100%. You should measure between C and GND a voltage of max. 3.3V (of it's less than 3.3V the max voltage of C is limited). Turn on the night light and measure the min/max voltage/setting of the nightlight. Turn the light on/off. Measure the voltage at the standby pin.

7. As soon you know your device very well, we can replace the white daughter board. I assume a lot of pads aren't connected. If we find pads/connections without knowing the purpose we are doomed. ;-)

splitatom44, 2021-04-04:
Excellent guide, next weekend when I have time I'll start poking around

lazzzrus, 2021-04-21:
Hello Syssi!

Thank you for your wonderful work!

I have similar light. 
So i can see output on UART. Miio net, miio monitor etc. It reacts when i change settings in yeelight app i.e.  "new britght 67".
Should i look for particular messages?

syssi, 2021-04-22:
You can identify a PWM signal if you measure the voltage of a GPIO. If you change the brightness the voltage should change.

Examples:
Warm white brightness 100%: 3.2V
Warm white brightness 50%: 1,5V
Warm white brightness 1%: 0,2V

lazzzrus, 2021-04-25:
Thank you!
So the reason was mediatek doesn't output pwm signal when is not connected to the mains.
Now i've managed to find cold and warm leds.

Is there way to disable mediatek without displacing it?
i'd want to connect esp32 to test everything. And only displace old MCU after everethyng is working.
Tried to connect start with the groud but no effect.

lazzzrus, 2021-04-25:
So these pins i discovered so far:
Anten	 -
Anten	 -
-              	 -
RX	         -
TX	         start
-	         -
CldLED	 -
WrmLED	 -
-	         -
-	         -
-	         G
-	         3v3
G	         3v3
G	          -

lazzzrus, 2021-04-25:
nightlight is very low warm leds. What the nightlight pin does and how to find it?

Same with RGB backlight. I'm not familiar with how it works yet. Should it be 3 different pins?

syssi, 2021-04-26:
The nightlight pin is a PWM pin too. It drives the warm white LEDs in a very low brightness range.

The RGB backlight are 3 PWM pins one per color.

syssi, 2021-04-26:
I prefer to remove the Mediathek MCU before assembling the ESP32 because it will be hard to remove the power supply from the Mediathek MCU only.

lazzzrus, 2021-04-26:
Hi Sebastian!
Thanks! 
So the pinout now looks like this:
[image]

What else should be considered before desoldering Mediatek?

Your yaml contains "power_supply" pin. Is it necessary or it could be not present in my device?

syssi, 2021-04-27:
At some devices a GPIO ist used (Low vs. High) to turn the LED driver on and off. Please check all remaining pins while turning the lamp on and off. Do you see a GPIO flipping between 0 and 3,3V?

lazzzrus, 2021-04-27:
Well, now it looks this way:
[image]

5 pins are always high but they are not connected to 3.3pin
4 pins are always low

1 pin between warm and nightlight is low when turned off and 1.55V when turned on
1 pin  between night and red is 0,87V when turned of an 1,55V when turned on.
What these two pins could be meant for?

Also i provided 3,3V to cold and warm LED pins when the light was turned off and the LEDs worked. 
Does it mean that the driver is always on and all we need is just to provide PWM output?

syssi, 2021-04-27:
> 5 pins are always high but they are not connected to 3.3pin
> 4 pins are always low

Just keep this in mind. May be the behaviour/state needs to replicated at the esphome firmware.

> 1 pin between warm and nightlight is low when turned off and 1.55V when turned on
> 1 pin between night and red is 0,87V when turned off an 1,55V when turned on.
> What these two pins could be meant for?

I don't know. 

> Also i provided 3,3V to cold and warm LED pins when the light was turned off and the LEDs worked.
> Does it mean that the driver is always on and all we need is just to provide PWM output?

Good idea. I think you are safe now. :-)

lazzzrus, 2021-04-27:
Finally got it working!
Sebastian, thank you for the help!

Issues so far:
1. Noise in night mode. May be i use not correct frequency for the LEDC component.  It's 1220Hz now.
2.  Need to make least voltage about 0,67V for main light.  Don't know how to do it yet. So the light can disappear when i'm changing color temperature or move lovelace scale to thee left.
3. Don't know why, but blue pin now turn the greern light) could make bridge when soldering. I'll fix it later.
4. Bluetooth dimmer probably won't work ever. There is open issue but nothing happens so far: https://github.com/esphome/feature-requests/issues/293
5. May be you can advice lovelace card for such ceiling lights? All-in-one card

syssi, 2021-04-28:
Could you provide a photo and your yaml as contribution?

jaddel, 2021-05-13:
> 2\. make least voltage about 0,67V for main light.  Don't know how to do it yet. So the light can disappear when i'm changing color temperature or move lovelace scale to thee left.

try these frequency settings to get ride of the noise. 

`output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    frequency: 4882Hz

  - platform: ledc
    pin: GPIO21
    id: output_cold
    frequency: 4882Hz

  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    frequency: 9765Hz`

jaddel, 2021-05-14:
I have the same light an was already thinking for a while to try this too. My soldering skills or the tools I where using have been simply not enough for the job. However I manged to restore full functionality except the Bluetooth remote. 

There you go. 
@syssi use any of the files for contribution if you want. 

https://github.com/jaddel/ESPHome-Configurations/tree/master/Devices/Ceiling%20Yeelight%20YLXD05YL

syssi, 2021-05-15:
@jaddel I've updated my ceiling lamps to esphome `1.18.0b3`. In the past there where connection dropouts at the history. With 1.18.0 the connection is rock solidsolid.

lazzzrus, 2021-05-15:
[image]
Can't say that my soldering skills are good as well. But it works.

lazzzrus, 2021-05-15:
> try these frequency settings to get ride of the noise.
>
> `output:
>
> * platform: ledc
>   pin: GPIO19
>   id: output_warm
>   frequency: 4882Hz
> * platform: ledc
>   pin: GPIO21
>   id: output_cold
>   frequency: [...]

Thank you, i will!

lazzzrus, 2021-05-20:
BLE-monitor custom integration now supports Yeelight dimmer and remote!
https://github.com/custom-components/ble_monitor#supported-sensors

Could you please advice how to set the mediatek controller into pairing mode? power off/on for 5 times doesn't work.
I need to add old controller to mihome app in order to get BLE encryption key.

syssi, 2021-05-20:
@lazzzrus The remote control can be used without any encryption key. The encryption key of the dimmer can be retrieved by a modded mihome app version. Do you want to give it a try? (No OEM firmware / Mediatek required)

lazzzrus, 2021-05-20:
i do for sure.
Actually i have both remote and dimmer.
The dimmer is not visible by miihome mod.
Remote is visible but returns error when try to pair with mihome mod.

Now i've managed to get mediatek back  into mihome mod, but no pairing .txt is available yet...

lazzzrus, 2021-05-20:
Well, miio returns another error:
C:\Users\user>miiocli device --ip <IP> --token <TOKEN> raw_command ble_dbg_tbl_dump '{"table":"evtRuleTbl"}'
Running command raw_command
Error: {'code': -9999, 'message': 'user ack timeout'}

syssi, 2021-05-21:
I assume at the daughterboard of your ceiling lamp is a small eeprom connected via i2c which is used to store the key. As long as your mediatek uC isn't back in place it's unable to retrieve the key from the eeprom.

I've used this modded mihome app: https://github.com/custom-components/ble_monitor/issues/289#issuecomment-844014615

Steps to pair the dimmer and retrieve the token:

1. Uninstall the original MiHome app
2. Install the modded APK (unknown sources must be enabled)
3. Login (use the region india as location/cloud/storage bucket)
4. Go to the settings of your android device (settings -> app -> mihome app) and make sure the app has permissions to write to the internal storage and to use the "phone permission". The first permission is required to write files to yor smartphone. The second permission is required for BLE (I hope I didn't miss a permission here).
5. Use a file manager to take a look at your internal storage. Do you see a `vevs/` directory here?
  - If not: Close / kill the app and start it again. If the app has the permission the folder should be created now.
  - If yes: Goto the directory `vevs/` and create a subdirectory called `logs`: `vevs/logs/`.
5. Close / kill the app.
6. Open the app. Goto the scanner/radar symbol at the top right corner and "scan" your environment for BLE advertisements.
7. Press and rotate the dimmer to provide some traffic
8. The dimmer should show up at the list of discovered devices.
9. Click on the dimmer to start the pairing.
10. If step 9 (pairing) doesn't complete start again at step 5. I had to disable bluetooth at my smartphone once and restart the app. If the pairing is successful it's super fast (3-6 seconds). 
11. Take a look at the internal storage. You should find a file called `pairings.txt` now. Alternatively use the https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor to login into the cloud and retrieve all devices from region `i2` (india).

It should look like this:

```
$ python3 token_extractor.py 
Username (email or user ID):
***********************
Password:
***********************
Server (one of: cn, de, us, ru, tw, sg, in, i2) Leave empty to check all available:
i2

Logging in...
Logged in.

Devices found for server "i2":
   ---------
   NAME:     小米无线开关2
   ID:       blt.7.16mdp9hbd0g00
   BLE KEY:  ababababababababababababFFFFFFFF
   TOKEN:    efefefefefefefefefefefef
   MODEL:    yeelink.remote.ctrl
   ---------

Press ENTER to finish
```

The key `abababababababababababab` is the key you are looking for.

lazzzrus, 2021-05-21:
Dear Sebastian! 
Thank you for so comprehensive guide!

I've tried two different smartphones, but both still can't pair with remote. 
Reinstalled several times. Rebooted phones, turned bluetooth on/off...
Dimmer is not visible at all. I'll keep trying with different phones. But it doesn't look good so far(

Now i even think about resoldering mediatek back. To get control over bluetooth devices.

Another guess: i've unpaired remote in mihome app. But not sure the remote "knows" about that. 
Pressed "M" and OFF buttons together as well...

syssi, 2021-05-21:
If I remember correctly the modded mihome app can be used to pair the **dimmer** only. @rezmus Do you agree?

Do you know the instructions of the ble monitor? May be I missed an important step to press the pairing button of the dimmer:

https://github.com/custom-components/ble_monitor/blob/master/faq.md#how-to-get-the-mibeacon-v2v3-encryption-key

In my case I had to start, stop and re-install the app multiple times. At the end everything went well. The BLE keys of the devices doesn't change over time. I printed the key to the back of my dimmers now ;-)

rezmus, 2021-05-21:
app can pair both dimmer and remote, for sure the one bought separately pid 339. not sure if light bundle remotes can be paired, depends on pid. do you have proper version of the app? click + and go to household security. scroll down to remote control, view more. you should see remote/dimmer icons there. i paired both with the app recently without issues.

bt device does not know if it's paired or not. it just keep sending adv encrypted with constant key. any device arround which knows algo and key can decrypt.

rezmus, 2021-05-22:
@lazzzrus if you still have no luck with the app here's python script to auth device and read beaconkey directly (use pid 950 for dimmer and 339 for remote).

https://github.com/rexbut/mikettle/blob/master/get_beacon_key.py

syssi, 2021-05-22:
@lazzzrus @rezmus It looks like there is another solution to retrieve the BLE key now: https://github.com/rexbut/mikettle/blob/master/get_beacon_key.py

rezmus, 2021-05-22:
@syssi look at previous comment ;)

syssi, 2021-05-22:
You are fast as hell! :-)

lazzzrus, 2021-05-22:
Still no luck with Android. The only difference it now sees the dimmer. But can't pair as well. Thank you guys! Will try a python!

lazzzrus, 2021-05-24:
This thing looks cursed.
I've installed two instances of debian in virtualbox on two different hosts, also raspbian in raspberry pi.
Always get the same result:
```
# python3 get_beacon_key.py F8:24:41:C3:37:36 950      Activate pairing on your 'F8:24:41:C3:37:36' device, then press Enter:
Connection in progress...
Traceback (most recent call last):
  File "get_beacon_key.py", line 144, in <module>
    main(sys.argv)
  File "get_beacon_key.py", line 141, in main
    get_beacon_key(mac, product_id)
  File "get_beacon_key.py", line 96, in get_beacon_key
    peripheral = Peripheral(deviceAddr=mac)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 391, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 439, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral F8:24:41:C3:37:36, addr type: public

```

jinhong-, 2021-09-03:
@lazzzrus I also have a mediatek board on my yeelights and the board layout looks the same. So far i managed to desolder the mediatek board. It is not clear to me what is the pinout i should be using. Is this the correct pinout?
https://github.com/syssi/esphome-yeelight-ceiling-light/issues/10#issuecomment-827154861

syssi, 2021-09-04:
[image]

I hope this helps a bit.

jinhong-, 2021-09-04:
Amazing. Thanks @syssi!

jinhong-, 2021-09-13:
I just received my ESP32 board I have more questions. In the initial discussions it was mentioned that the following pins need to be identified:
- START
- 3.3V
- GND
- RXD
- TXD
- C (Cold white), PWM signal
- W (Warm white), PWM signal
- Night light (if supported), PWM signal
- Standby / Power supply (if supported)

The following I understand how I should be mapping/connecting
- 3.3V
- GND
- C (Cold white), PWM signal
- W (Warm white), PWM signal
- Night light (if supported), PWM signal

The following I understand what they are, but do not understand how and why we need to connect them. Do we connect them to the TX and RX of the ESP32 or use other gpio pins
- RXD
- TXD

The following I don't know what they are for, or which pins on the ESP32 I should connect them to
- START
- Standby / Power supply (if supported)

I am also assuming R, G, B pins are more for lights that have RGB support, which mind doesn't so thus I don't have to connect them

any thoughts @syssi? Thanks in advanced :D

syssi, 2021-09-13:
You are right: RXD and TXD isn't important. Both pins are required the flash esphome once. You don't need to solder these pins to the daughter board. START is also not important because it's the RESET pin of the Mediatek micro controller. At the beginnen I was unsure about the purpose of the pin. STANDBY is important if you ceiling lamp doesn't turn on if you apply a PWM to C or W. The STANDBY pin turns the power supply of LED driver board on/off (on some devices).

jinhong-, 2021-09-13:
Gotcha. I'll give it a try and see what i get

jinhong-, 2021-09-13:
@syssi It works great. Just got everything wired in. One thing - the light seems to turn off before hitting 1% brightness. How do I adjust it so that it will stay on at 1%?

syssi, 2021-09-13:
This is a well known issue: https://github.com/syssi/esphome-yeelight-ceiling-light/issues/15#issuecomment-850946633

If you are able to solve/improve it please share your configuration!

jinhong-, 2021-09-13:
For others to reference this post in the future, this is the configuration I used that worked for me
```
#Variables
substitutions:
  device_name: main-bedroom-light
  friendly_name: Main Bedroom Light

esphome:
  name: ${device_name}
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
# logger:
api:
ota:
  password: !secret ota_password

output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    frequency: 4882Hz
    min_power: 0.07
    zero_means_zero: true

  - platform: ledc
    pin: GPIO21
    id: output_cold
    frequency: 4882Hz
    min_power: 0.07
    zero_means_zero: true

  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    frequency: 9765Hz

light:
  - platform: monochromatic
    name: "${friendly_name} Nightlight"
    id: night_light
    output: output_nightlight
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: ceiling_light
  - platform: cwww
    name: "${friendly_name}"
    id: ceiling_light
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K
    gamma_correct: 0
    constant_brightness: true
    on_turn_on:
      - light.turn_off: night_light
```

syssi, 2021-09-13:
Cool! I didn't know `zero_means_zero`.

wf1nder, 2021-10-14:
> May be you should try to pull-down/pull-up the standby pin once. Just to make sure the lamp doesn't turn off in one of the cases. It's possible the LED driver / power supply runs permanently now and it would be better to use the standby pin.

Have anyone found out which pin is 'standby' on mediatek board?
Anyway, I have already installed the lamp on the ceiling, and I don't want to climb to it yet :) So far everything is working fine without any failures, including when I powering off/on lamp. But I will keep in mind in case something goes wrong.

Notrial, 2022-01-28:
> ndby pin turned out to be unnecessary. And here my config:
> 
> ```
> #Variables
> ```

Any updates on how it's working so far? Is it dimming nicely?
I have three of those, just want to be sure before I proceed :)

Notrial, 2022-02-02:
> Have anyone found out which pin is 'standby' on mediatek board? Anyway, I have already installed the lamp on the ceiling, and I don't want to climb to it yet :) So far everything is working fine [...]

Any updates on how it's working so far? Is it dimming nicely?
I have three of those, just want to be sure before I proceed :)

jarivd, 2022-02-03:
> Any updates on how it's working so far? Is it dimming nicely? I have three of those, just want to be sure before I proceed :)

I just finished doing this with my ylxd01yl (ceiling1) and it's dimming perfectly, also got the remote to work over bluetooth using https://github.com/dentra/esphome-components/tree/master/components/miot_ylyk01yl and with the bindkey I extracted using the mi home app that's linked somewhere above in this thread (https://ru.kapiba.ru/mihome/files/public/others/MiHome_6.5.700_63911_vevs_dimmer.apk).
I believe this: https://github.com/dentra/esphome-components/tree/master/components/miot_ylxx0xyl_pair also works to get the bindkey, though. Might be easier. I basically used the same yaml setup as wf1nder did, just included the bluetooth remote setup in there as well.

jarivd, 2022-02-03:
Thank you all guys, I managed to get my YLDX01YL to work perfectly thanks to you. I set up my bluetooth remote a little different though. I used this repo https://github.com/dentra/esphome-components/tree/master/components/miot_ylxx0xyl_pair to get the bindkey off of the remote that came with my light and from the same author I used this repo https://github.com/dentra/esphome-components/tree/master/components/miot_ylyk01yl to make it work on the esp32 that's controlling the lights. 
I tried the ble_monitor setup linked above as well, but I believe it only works using the bluetooth on the home assistant itself (or I might have done something wrong), but the room I use the remote in is too far away for it to have a stable connection.
I used pretty much the same yaml as wf1nder, just with the added automations for when a button on the remote is pressed, and it's working great so far!

spn91, 2022-05-16:
Hello,
I somehow managed to remove the mediathek device from my YLXD05YL (was very hard with my beginner tools). If I compare my board with the screenshots in this issue, it seems that mine is different? Can someone help me please? :) i know that I have to clean the board first, but my bottle with isopropanol is empty..

[image]

syssi, 2022-05-16:
I assume your biggest issue is the shorting of the two pads at the bottom left of the photo. ;-) Could you provide some more and high-res photos of your mainboard?

spn91, 2022-05-16:
Which pads do you mean? The ones which are completely „brown“? Is there something I can do to rescue the board? :(

Is Github compressing the photos? The last one should already be high-res. 

But here some more. 

[image]
[image]

syssi, 2022-05-17:
It's fine now. I was talking about the two connected pins next to C39.

syssi, 2022-05-17:
Could you provide a photo of the mediatek board? I would like to make sure this board hasn't changed.

spn91, 2022-05-17:
I have found this picture, it looks like this is the same layout: 
[image]

Do you know if this is a working configuration? I would connect 3,3V and GND to the same row where the START pin is, but all the others to same as shown in the picture.

spn91, 2022-05-17:
[image]

[47 more replies]