syssi/esphome-yeelight-ceiling-light #11: [Compiling Error] Got the unrecognized status code '404' when downloaded arduino-lib

jaddel, 2021-05-12:
`--------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/Jason2866/esp32-arduino-lib-builder/raw/framework-arduinoespressif32/framework-arduinoespressif32-release_v3.3-solo1-bd65eb8d1.tar.gz
Error: Got the unrecognized status code '404' when downloaded https://github.com/Jason2866/esp32-arduino-lib-builder/raw/framework-arduinoespressif32/framework-arduinoespressif32-release_v3.3-solo1-bd65eb8d1.tar.gz`

Running the latest esphome in HomeAssistant i'm getting this error trying to upload / update the device.

syssi, 2021-05-12:
This download isn't available anymore. I will lookup for an alternative.

syssi, 2021-05-13:
It looks like the solo1 core cpu is supported official now. I didn't test this change. Could you provide some feedback it's working or not? Thanks in advance!

closed 2021-05-13 by syssi

syssi, 2021-05-13:
Successfully tested. Works fine!

commit da0075ba (2021-05-13): Use release solo1 core1.0.6 (Closes: #11)

https://github.com/arendst/Tasmota/commit/9bd46df6007c3272049458406acd111da30a9e59#diff-ebebbfa1b5737117b01c43f27a1f78f7d2c75bf3bfa353b6a0d1818357b23205