syssi/esphome-yeelight-ceiling-light #13: Add Yeelight Rotating Dimmer (YLKG07YL, YLKG08YL) support

syssi, 2021-05-17:
cp. https://github.com/custom-components/ble_monitor/issues/289

gudszent, 2021-12-21:
do you have plan to resolve the ble reboot issue?

syssi, 2021-12-21:
I've replaced the single core ESP32 of one of my lamps with a dual core ESP32. At dual core ESP32 the `ble_monitor` is stable. As soon BLE and WiFi runs on the same core the application crashs. I don't know how to fix it. It's above my skills. ;-)

syssi, 2021-12-21:
Oh there is some progress missing here. I've extended the `ble_monitor` of ESPHome to support the remote controls. You can find an example configuration here: https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeerc_ylyk01yl.yaml

The problem: If you enable the BLE stack on an single core ESP32 (used in all Yeelight devices) the application crashs on boot-up. To be able to listen to some BLE traffic you have to replace the ESP32 with a dual core version.

gudszent, 2021-12-21:
Thank you, as I see the config, that not direct control the light, its send it to home assistant, than i guess the rule from HA control the esp. Am I right?

gudszent, 2021-12-21:
As I see, the only chance to turn off and on the wifi and ble :-)
https://github.com/espressif/arduino-esp32/issues/1756
So I think, this is the dead end :'(

syssi, 2021-12-21:
The Yaml is just an example. You could wire/cosume the events also in ESPHome directly.

commit 66f1b7a7 (2025-08-13): Add Yeelight Rotating Dimmer (YLKG07YL, YLKG08YL) support (#32)

Closes: #13

How to retrieve the required key:
https://github.com/custom-components/ble_monitor/issues/999#issuecomment-2102056708
closed 2025-08-13 by syssi