syssi/esphome-yeelight-ceiling-light #14: ESPHome-Instructions

chris-nite, 2021-05-28:
Thank you so much @syssi for your great work on unchaining these device from chinese cloud servers!
I would love to get my meteorite in esphome, but right now, i am a little bit struggling with your instructions.
I got a USB-TTL-Dongle ready up for some flashing action, but i actually don't understand how i can hold down the state of the GPIO01-Pin on the ESP. Do i need to bridge something to get it in that state?
Would you be so kind and give me some instructions how to get the esp32 in the proper state, to flash it through the dongle with esphome? The latter part i already know from to epaper-esp32-devices, but i'm just struggling with the part of preparing the esp32-board.
Thank you in advance and best wishes
Chris

syssi, 2021-05-28:
No problem. It provide some instructions later the day.

syssi, 2021-05-28:
How to flash the meteorite:

1. Disassemble the device to have proper access to the daughterboard.
2. Solder 4 (if possible 5) wires to the board (3.3V, RX, TX, GND, GND): https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/yldl01yl/007.jpg
3. The fifth wire is to have propery access to GND. You could use the second test point on the left if you don't want to solder two wires to the same test point (on the right).
4. Make sure the dongle isn't plugged in.
5. Attach RX , TX, GND and 3.3V to your USB-TTL-Dongle. Please be aware: The dongle won't provide enough power to drive/boot the ESP32 and connect to your WiFi. If you want to play around with the serial output of the stock firmware you've to provide the voltage from another source providing enough current. Just for flashing the device (booting into the flash mode) the power of the dongle could/should be enough.
6. Connect the fifth wire (GND) to the test point IO0. This is called: "Pull IO0 down". If GPIO0 of an ESP32 is pulled down and you connect power to the device the microcontroller goes into the "download mode" (flash mode).
7. Plug the dongle into the PC (make sure IO0 is pulled down). You could also have the dongle always plugged into your USB port and simply connect 3.3V to the board (while pulling IO0 down). If you look at the serial traffic at the same time you would see the this message:

```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))

waiting for download
```

8. Now you've entered the download mode and the ESP32 can be flashed via esphome. 
9. If you are done please don't expect the ESP32 is able to connect to your WiFi (after reboot). You have to remember: Your dongle doesn't provide enough current.

chris-nite, 2021-05-29:
Thank you very much for the comprehensive guide.
Today i disassembled the light and soldered the pins on the mentioned points. I also soldered a wire on the IO0, which i then held down to the GND near the TX and RX points. Unfortunately i am not able to get any serial output through putty and the COM-Port of the Dongle. Also i'm not able to flash the binary through the esphome-flasher, as it is unable in detecting the esp chip.
It seems the ESP doesn't go into Download mode and i'm currently at loss here, what might be wrong.
Do i need a special configuration of the usb-dongle (it is an AZDelivery UART-TTL USB Adapter with CH340G Chipset on Standard Windows-Settings)?
As the ESP itself doesn't show any activity (a LED for instance), i'm also not sure, if it powers up at all.
Do you have any ideas?

syssi, 2021-05-29:
Did you try to swap TX und RX?

chris-nite, 2021-05-29:
Hah! You're my hero. Swapping TX and RX helped. Funny stuff. Thanks!

closed 2021-06-03 by syssi

lolopepes, 2024-07-03:
> How to flash the meteorite:
> 
> 1. Disassemble the device to have proper access to the daughterboard.
> 2. Solder 4 (if possible 5) wires to the board (3.3V, RX, TX, GND, GND): https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/yldl01yl/007.jpg
> 3. The fifth wire is used to have easy access to GND. You could use the second test point on the left if you don't want to solder two wires to the same test point (on the right).
> 4. Make sure the dongle isn't plugged in.
> 5. Attach RX , TX, GND and 3.3V to your USB-TTL-Dongle. Please be aware: The dongle won't provide enough current to drive/boot the ESP32 and connect to your WiFi. If you want to play around with the serial output of the stock firmware you've to provide the voltage from external power supply providing enough current. Just for flashing the device (booting into the download mode) the power of the dongle could/should be enough.
> 6. Connect/press the fifth wire (GND) to the test point IO0. This is called: "Pull IO0 down". If GPIO0 of an ESP32 is pulled down and you connect power to the device the microcontroller goes into the "download mode" (flash mode).
> 7. Plug the dongle into the PC (make sure IO0 is pulled down). You could also have the dongle always plugged into your USB port and simply connect 3.3V to the board (while pulling IO0 down). If you look at the serial traffic at the same time you would see the this message:
> 
> ```
> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> 
> waiting for download
> ```
> 
> 8. Now you've entered the download mode and the ESP32 can be flashed via esphome. As soon the device as entered the download mode (1 second after powering up) you don't have to pull down GPIO0 any longer.
> 9. If you are done please don't expect the ESP32 is able to connect to your WiFi (after reboot). You have to remember: Your dongle doesn't provide enough current. Add an external power supply or reassemble the device. In future the device can be flashed via OTA.

Hi you,
Recently flashed a Meteorite lamp with esphome but now is in boot loop.
The logs sais this and i cant enter on bootmode with pull IO0 down

Can you help me?

[08:41:21]ets Jun  8 2016 00:22:57
[08:41:21]
[08:41:21]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:41:21]configsip: 0, SPIWP:0xee
[08:41:21]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:41:21]mode:DIO, clock div:2
[08:41:21]load:0x3fff0030,len:1184
[08:41:21]load:0x40078000,len:13132
[08:41:21]load:0x40080400,len:3036
[08:41:21]entry 0x400805e4
[08:41:22]E (626) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
[08:41:22]E (626) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig
[08:41:22]
[08:41:22]abort() was called at PC 0x40082cfd on core 0
[08:41:22]
[08:41:22]
[08:41:22]Backtrace:0x4008374d:0x3ffe3b800x4009310d:0x3ffe3ba0 0x400986d1:0x3ffe3bc0 0x40082cfd:0x3ffe3c40 0x400792ba:0x3ffe3c90  |<-CORRUPTED
[08:41:22]
[08:41:22]
[08:41:22]
[08:41:22]
[08:41:22]ELF file SHA256: 0000000000000000
[08:41:22]

syssi, 2024-07-03:
Did you change the YAML configuration? Which ESPHome version did you use?

You should be able to enter the boot mode nevertheless. This should work always. Please double check your wiring.

lolopepes, 2024-07-03:
> Did you change the YAML configuration? Which ESPHome version did you use?
> 
> You should be able to enter the boot mode nevertheless. This should work always. Please double check your wiring.

I used the yaml for ceiling.10, and the wires are soldered. If a pull io0 log not show anything and if a i start without io0 pull the log show what i pasted before.

Any idea to start bootmode? I used esphome web flash for first flash.

syssi, 2024-07-03:
Please pull GPIO0 and try to flash the ESP another time. Does the ESP accept the upload? It should!

lolopepes, 2024-07-03:
> Please pull GPIO0 and try to flash the ESP another time. Does the ESP accept the upload? It should!

I tried many time with same result.
Can i flash with other software with new bootloader or any else? like ESP32 FLASH DOWNLOAD TOOL?

syssi, 2024-07-03:
Yes. You can flash the ESP with any software/programmer you want.

lolopepes, 2024-07-03:
I tried, but no lucky, bootmode not work. You know any other mode to boot with bootmode? any software can do this?

syssi, 2024-07-03:
No. AFAIK pulling GPIO0 is the only way to boot the ESP into the download mode. Could you provide some photos and describe what do you do exactly?

lolopepes, 2024-07-04:
Hi Sebastian,

Pulling io0 don't work bootmode, i tried serveral times and nothing work.
I'm thinking desolder the ESP32 and solder a new one programed previously.

Thanks for all.