syssi/esphome-yeelight-ceiling-light #15: Meteorite

chris-nite, 2021-05-29:
Hi @syssi , thanks again for your fast help in getting the meteorite up and running with esphome.
Finally a functioning and responsive light! :-)
When using the light i today found out, that when the main light is used with a low brightness and then changing the white tones somewhere to the middle, the presented white tone is very warm or cold (depending if you move the slider a bit to the right or left). Changing tones then are done very aprubtly by the light, while very shortly flashing in the right (mixed) color. There seems to be something off, because if you improve the brightness by quite a bit, changing white tones go very smoothly and are correctly mixed (through the ww and cw leds).
Attached you'll find my yaml-file.
Anyone got the same issues on lower brightness levels?

```
substitutions:
  name: yeelight_light_ceiling10

esphome:
  name: ${name}
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.0.0
    platform_packages: tasmota/framework-arduinoespressif32 @ 3.10006.210420

wifi:
  ssid: xxxx
  password: xxxx
  fast_connect: true

ota:
api:
logger:

output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    power_supply: power
#    min_power: 0.13
    max_power: 0.82
  - platform: ledc
    pin: GPIO21
    id: output_cold
    power_supply: power
#    min_power: 0.13
    max_power: 0.82

  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    power_supply: power

  - platform: ledc
    pin: GPIO33
    id: output_red
    power_supply: power
  - platform: ledc
    pin: GPIO26
    id: output_green
    power_supply: power
  - platform: ledc
    pin: GPIO27
    id: output_blue
    power_supply: power

power_supply:
  - id: power
    pin: GPIO22
    enable_time: 0s
    keep_on_time: 0s

light:
  - platform: monochromatic
    name: "${name} night light"
    id: night_light
    output: output_nightlight
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: ceiling_light
  - platform: cwww
    name: "${name} ceiling light"
    id: ceiling_light
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: night_light
  - platform: rgb
    name: "${name} ambient light"
    red: output_red
    green: output_green
    blue: output_blue
    gamma_correct: 0
    effects:
      # Use default parameters:
      - random:
      # Customize parameters
      - random:
          name: "Slow Random Effect"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "Fast Random Effect"
          transition_length: 4s
          update_interval: 5s
      - pulse:
          name: "Fast Puse"
          transition_length: 0.5s
          update_interval: 0.5s
      - pulse:
          name: "Slow Puse"
          # transition_length: 1s      # defaults to 1s
          update_interval: 2s
      - strobe:
          name: "Green Strobe"
          colors:
            - state: True
              brightness: 100%
              red: 0%
              green: 100%
              blue: 0%
              duration: 500ms
            - state: False
              duration: 250ms
      - strobe:
          name: "Red Strobe"
          colors:
            - state: True
              brightness: 100%
              red: 100%
              green: 0%
              blue: 0%
              duration: 500ms
            - state: False
              duration: 250ms
      - strobe:
          name: "Blue Strobe"
          colors:
            - state: True
              brightness: 100%
              red: 0%
              green: 0%
              blue: 100%
              duration: 500ms
            - state: False
              duration: 250ms
      - strobe:
          name: "Police Strobe"
          colors:
            - state: True
              brightness: 100%
              red: 0%
              green: 0%
              blue: 100%
              duration: 500ms
            - state: True
              brightness: 100%
              red: 100%
              green: 0%
              blue: 0%
              duration: 500ms
      - flicker:
```

syssi, 2021-05-29:
Please remove all `gamma_correct: 0` lines, reflash and try again. ;-)

chris-nite, 2021-05-30:
I tried that out and now the light only turns on, when i pull the brightness lever to about 40%. Interestingly the white tones still occur that flashing behavior when pulling it to about 50%. It seems the light tries to mix the warm and cold diodes, suceeds for about half a sec and then falls back to colder or warmer tones.
Might the frequency setting change a thing?

syssi, 2021-05-30:
I've disabled (value: 0) the gamma correction in the past because of this behavior / issue: https://github.com/esphome/esphome-core/issues/190

I will try to reproduce the issue at my device to get a better understanding. I must admit I don't use the color temperature slider.

Could you try to play around with the frequency setting? I think it could improve the situation (especially if driving the light with more brightness is working fine).

syssi, 2021-05-30:
Take a look at this issue for some additional informations:

https://github.com/syssi/esphome-yeelight-ceiling-light/issues/8

I think the usability of every supported device / esp32 driven yeelight can optimized easily by playing around with `gamma_correct`, `min_power`, `max_power` and the `frequency` setting. May be some custom code is required to fix/improve the lower brightness limit (brightness=1).

chris-nite, 2021-06-09:
I'm sorry, that i didn't give you a response yet. I really appreciate your support and will try out your suggestions on the coming weekend. Can i freely test out these settings like frequency/power etc. on the device itself without the risk of destroying it?

rafasanz, 2022-10-10:
With the actual yaml if light is under 13-15% can happen few situations:

- under 13% always turn off
- on 14% leds on are always white cold (sometimes flickering)
- on 15% normal function for cold and warm light

I have uncomment line 29 & 35 and now it's looking that everything is working fine.

Thanks and nice work!!

ultrazauberer, 2023-03-28:
I just want to say I have the same issue with the stock firmware. Changing the color of the white Mainlight on low brightness causes color flickering. Feels like the old light bulbs in the office.

So the ESPhome firmware could fix this problem according to the last post?

Also on of my yeelight arwen 550 forgets all the settings and the link to my Xiaomi account. But I barely use this lamp. Could ESPhome here also improve the behavior?

syssi, 2023-03-28:
> Also on of my yeelight arwen 550 forgets all the settings and the link to my Xiaomi account. But I barely use this lamp. Could ESPhome here also improve the behavior?

There is an EEPROM on the daughter board to store some key material and may be account numbers. This EEPROM isn't used if you flash the ESPHome based firmware so the chance is high to improve the usability/stability of your device.

ultrazauberer, 2023-03-28:
> There is an EEPROM on the daughter board to store some key material and may be account numbers. This EEPROM isn't used if you flash the ESPHome based firmware so the chance is high to improve the [...]

Thanks for you fast respond. I will definitely try it. Something is coming to my mind:
If I use ESPhome on the yeelight, could I connect a DHT22 and read humidity and temperature (one big disadvantage: I guess only when the lamp is on)?

syssi, 2023-03-28:
> If I use ESPhome on the yeelight, could I connect a DHT22 and read humidity and temperature (one big disadvantage: I guess only when the lamp is on)?

Yes, the number of possibilities is huge here. :-) In best case the device is powered always.

PsychoRS, 2023-11-07:
Hi, I think I've found the solution to 100% replicate the stock firmware behaviour.

First of all, I've recalculated min_power and max_power using the voltage info here. So, with my i-dont-know-if-correct-but-it-works calculations, max_power for the warm output must be 0.77 (if 0 is 0 and 1 is 3.3v, so 2.55v must be 0.77), and 0.76 for the cold output.

So:

`output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    power_supply: power
    min_power: 0.13
    max_power: 0.77
    frequency: 1220Hz
  - platform: ledc
    pin: GPIO21
    id: output_cold
    power_supply: power
    frequency: 1220Hz
    min_power: 0.13
    max_power: 0.76`

With this settings, my brightness slider is working perfect (no more turning off at 12-13%, no more "no more brightness" from 9x% up". Then I found that, with that settings, if you turn off the light when at low brightness, it flicks strangely, as @rafasanz said.

The solution? For me was just using the zero_means_zero parameter.

So:

```
output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    power_supply: power
    min_power: 0.13
    max_power: 0.77
    frequency: 1220Hz
    zero_means_zero: true
  - platform: ledc
    pin: GPIO21
    id: output_cold
    power_supply: power
    frequency: 1220Hz
    zero_means_zero: true
    min_power: 0.13
    max_power: 0.76
```

syssi, 2023-11-07:
Awesome! I will give it a try!

closed 2024-09-13 by syssi

helgek, 2024-10-14:
@PsychoRS Thank you so much, my meteorite now also produces really perfect results!!