syssi/esphome-yeelight-ceiling-light #21: Can we switch the esp32 and keep the original?

jmvaz, 2021-09-06:
@syssi I have one ceiling 20 and I wonder if I can flash an esp32 mini and wire it directly with jumper cables and keep the original module. What do you think? Thks!

syssi, 2021-09-06:
Could you explain the goal? Do you want to keep the stock firmware as fallback? Do you want to use bluetooth but do you try to avoid to desolder the single core esp32?

jmvaz, 2021-09-06:
The goal was to keep the original chip untouched. But I did a dump of the firmware, and if I need can rollback. Thanks a lot for your solution. I was really freaking out with the shitty software from Xiaomi. Esphome is now working very smoothly... :) Thanks @syssi

closed 2021-09-06 by jmvaz

syssi, 2021-09-07:
I assume piggybacking a second ESP is only possible if you are able to remove the power or don't boot from the one you won't use. I recommend to replace the firmware.

chex01, 2022-01-13:
> I did a dump of the firmware

Hello, @jmvaz !
If you don't mind, send a dump of the factory firmware, thanks!