syssi/esphome-yeelight-ceiling-light #22: Yeelight Arwen 450C (YLXD013-B)

BotoX, 2021-09-29:
Yeelight Arwen Ceiling Light 450C
YLXD013-B
https://en.yeelight.com/product/1383.html
"50W UFC-N Mainboard V02"
"P777 LED DRIVE BOARD"
"WIFI UFC V01"
"UFC RGB V1.0"

Have two of those and am kinda annoyed that they can't get as dark as my older yeelight ceiling lights.
So here are some pictures of it, seems very similar to other supported ones.
https://cloud.botox.bz/s/eFHFLfKRykkm4bJ

Seems to come with dual core ESP32-WROOM-32D (ESP32-D0WD)

Also this project should have a few more infos in the readme:
What works/What doesn't? (I really would like to use the BT remote for example)
How to flash/update?
Revert to original firmware?

syssi, 2021-09-29:
1. The bluetooth remote is supported if you replace the ESP32 to a dual core version
2. Attach RX, TX, GND and VCC to the ESP32. Pull-down the GPIO0 on boot-up. Flash the esphome yaml.
3. I wasn't able to dump the OEM firmware. At the moment there is no way back.

soulcola, 2021-11-19:
Is it working with esphome?

syssi, 2021-11-20:
The device is esp32 based. If somebody identifies the GPIOs it can be used with esphome. I could provide some instructions to get it done if you own a multimeter.

ChrisMuc, 2021-12-30:
@BotoX How did you open it? Can you tell me, where the clips are located and which tools you used to remove the cover.

@syssi What do you mean with "The bluetooth remote is supported if you replace the ESP32 to a dual core version"? It supports WIFI and Yeelight's bluetooth remote with original firmware. Maybe it's already a dual-core ESP32 inside. Or did someone figure out, that it's just a single-core?

syssi, 2021-12-30:
In general yeelight safes money by using a single-core ESP32. I haven't seen a yeelight device yet equipped with a dual-core ESP32 by factory. The stock firmware is "optimized" to allow BLE and WiFi coexisting on a single core. ESP-IDF + ESPHome will crash on boot-time if you enable BLE. If you want to use BLE with ESPHome a dual-core ESP32 is required at the moment.

syssi, 2021-12-30:
> I wasn't able to dump the OEM firmware. At the moment there is no way back.

This isn't true anymore. You should try to dump the flash. I could provide some instructions.

stefan0001, 2021-12-31:
> This isn't true anymore. You should try to dump the flash. I could provide some instructions.

Hi @syssi  is there something special when creating a backup? 
Found some example for esptool: 
`python ./esptool.py --port /dev/ttyUSB4 --baud 115200 read_flash 0x00000 0x400000 backup.img`
Does this work for yeelight? 

Also thanks for the create work here!

syssi, 2021-12-31:
Yes! This command will work!

cp. https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/doc/flashing.md#make-a-backup-of-the-current-firmware

marrobHD, 2022-05-13:
Did anyone found gpio pins yet? I have the 550C model. If I'm not wrong it has specific LEDs for nightlight?

syssi, 2022-05-13:
If you provide some high-res photos I could try to help to identify the pins.

marrobHD, 2022-05-13:
I'll look into it at weekend and try to figure most of it out.

syssi, 2022-05-13:
You can find some other issues with details about how to identify the correct GPIOs at this project. Good look! Feel free to ask if something is unclear.

marrobHD, 2022-05-20:
[image]
[image]
[image]

syssi, 2022-05-21:
You could use a multimeter to measure the continuity between the RX, TX and the GPIO0 pin an the test pins (`TPx`) on the back of the board. If you can find test pins for RX, TX and GPIO0 it's easier to solder.

marrobHD, 2022-05-21:
I guess there arent any testpoins on the back. There are only 8 pins from the esp going to the mainboard.

syssi, 2022-05-21:
I'm talking about the back of the small daughter board not the LED driver board. All pins of the ESP should be soldered to the daughter board.

syssi, 2022-05-21:
`TP3` should be `GPIO0` f.e.

[image]

marrobHD, 2022-05-22:
I found the reason why it didn't work before. I had the wrong esp32 template in front of me. Pinout was different. This matches the esp32 on the board:
[image]

I dumped the firmware and will now try to flash ESPHome.

#### Test points WIFI UFC V01:
TP3 -> GPIO0
TP4 -> GPIO3, RXD0
TP5 -> GPIO1, TXD0
TP1 -> 3V3
TP2 -> GND

syssi, 2022-05-22:
Good job!

marrobHD, 2022-05-22:
ESPHome flashed. Everything works, but I also noticed that the network connection is more unstable than with original firmware.

syssi, 2022-05-22:
My yeelights are pretty stable (1 reconnect per day). Is your access point far away?

[image]

syssi, 2022-05-22:
This is my wifi config:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  use_address: 192.168.xxx.xxx
  power_save_mode: LIGHT
  reboot_timeout: 5min
```

marrobHD, 2022-05-22:
AP isn't too far away (-68dBm).
Nevermind, it stopped being unstable. 
[image]

syssi, 2022-05-23:
@marrobHD I would like to add your ceiling lamp to the list of supported device. Do you like to contribute your config yaml? Do you know the miio model name (something like `yeelink.light.ceilingXX`)?

marrobHD, 2022-05-23:
The miio model name is just `yeelight.light.ceilc`.
I used [this](https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceiling20.yaml)  configuration as base and it almost worked out of the box. Some adjustments were needed to get it working with esp-idf framework.
My config can be found [here](https://gist.github.com/marrobHD/8558c00e97070fcee1f1e5064943e89b).

closed 2022-05-23 by syssi

spn91, 2022-05-24:
I can confirm that this is also working for the Yeelight Arwen A2001. This is the same light, but without the RGB lights. It is identified as yeelink.light.ceil29.

syssi, 2022-05-25:
@spn91 Do you own the 50 or 60cm version? I'm asking because of the model identifier. So I can add an question mark to the identifier of the other version.

spn91, 2022-05-25:
It is the smaller version.

slimcdk, 2022-12-30:
This also works for the 550S. I have a dump of the original firmwares. Both 2.1.7_0009 and 2.1.7_0011 (this one from the OTA download aswell, but something still seem to be missing). 

I wish to be able to do OTA provision to ESPHome in the future, so I have network traffic captures of the devices as well as decoded mobile app source code, but have not gotten further as of today.

DenysShmyhin, 2023-01-02:
@slimcdk hi, could you pls share the original dumps for Ceiling Light 550S (YLXD013-A) I'm trying to recover the original firmware after a power outage. Now it is turned into factory mode and all solutions to turn it into default mode didn't work for me (after power on/off it becomes the same factory mode).Thanks in advance

maumaumaumaumau, 2023-02-03:
I have an issue with connecting the lamp correctly. 
I do exactly as described here: 
TP3 -> GPIO0
TP4 -> GPIO3, RXD0
TP5 -> GPIO1, TXD0
TP1 -> 3V3
TP2 -> GND

additionally I connect GPIO0 to GND, rest is as always: RXD0 -> TXD, TXD0 -> RXD and 3V3 -> 3V3, GND -> GND.

If I turn on the lamp with CH340G or FT232RL WITHOUT USB cable attached, the lamp goes into flash mode. But If I connect the USB Cable to either flashing devices, the lamp shorts (and is dead). What am I doing wrong?

syssi, 2023-02-03:
Please never power your ceiling lamp using 230VAC if it's open for flashing!! Just power the ESP using 3,3V. As soon the ESP is flashed remove all cables, close the cover and attach the lamp to the ceiling again.

maumaumaumaumau, 2023-03-12:
FYI: The fuse blew up, I changed it, flashed the lamp. Working like a charm.

syssi, 2023-03-12:
Good job!

helgek, 2023-04-15:
> @BotoX How did you open it? Can you tell me, where the clips are located and which tools you used to remove the cover.
>
> @syssi What do you mean with "The bluetooth remote is supported if you replace [...]

@BotoX @ChrisMuc I'd also be interested to understand how you can remove the cover. Information on this would be very much appreciated! I'm able to lift a bit the side using a plastic slider but I don't get beyond that.

vladimir1vladimirovich, 2023-09-27:
Hi there!

I connected to the Xiaomi YLXD013-C (esp32-woom-32d) lamp via UART and tried to upload the program via Arduino IDE, the upload happened, but the terminal said: **“running on single core variant of a chip, but app is built with multi-core support"** and esp kept rebooting all the time.

Please tell me how to flash this esp without using esphome firmware.

P.S. The lamp is planned to be used in OpenHAB.

syssi, 2023-09-27:
Please give it a try using ESPHome. Just avoid the `api` componnent and use the `mqtt` component to talk to your lamp instead. I assume OpenHAB supports MQTT too.

I don't know how to convince the Arduino IDE to enable the required single core build.

rdorsch, 2024-01-19:
Does anybody now if the LEDs in the lamp are controlled with a [WS2812B protocol](https://www.arrow.com/en/research-and-events/articles/protocol-for-the-ws2812b-programmable-led)? There are four wires to the lamp PCB, which is one more which is strictly needed fro WS2812B.

syssi, 2024-01-19:
@rdorsch Why do you think it's WS2812B? IMO the LEDs cannot be controlled individually. You have to control the complete string using PWM.

rdorsch, 2024-01-19:
I had a LED stripe with WS2812B. But you are right, looking at the PCB it even says VW+, VW-, VC+, VC- for the warm and cold white LEDs.

Myakishi, 2024-08-27:
Hi.Please share the factory firmware (dump)

helgek, 2024-08-27:
[fwbackup_Arwen_Bathroom.bin.zip](https://github.com/user-attachments/files/16770534/fwbackup_Arwen_Bathroom.bin.zip)

@Myakishi this is my backup file of the 450C

Myakishi, 2024-08-28:
Спасибо 28.08.2024, 01:58, "helgek" ***@***.***>***@***.*** this is my backup file of the 450C

PsychoRS, 2024-12-01:
Hi,

How do you dissassemble the full board of the Arwen 550S from the  case without breaking/desoldering the cable that goes to the leds itself?

fir3drag0n, 2025-02-21:
> Hi,
> 
> How do you dissassemble the full board of the Arwen 550S from the case without breaking/desoldering the cable that goes to the leds itself? I don't know how to remove the full board (including the ESP32 daughter board) from the lamp case to be able to flash the daughter board...

Did you solve it?

PsychoRS, 2025-02-21:
> Did you solve it?

Yes, I finally managed to flash it without removing that cable, just fiddling around and using some PCBite tools to keep the board in the needed position:

[image]

As you can see, the cable for the ambilight led strip is still connected.

slimcdk, 2025-02-21:
> Did you solve it?

Mine was a single cable that could be unplugged.

fir3drag0n, 2025-02-21:
Which variant do you own?

PR #38 (2022-05-23): Add YLXD013-B & YLXD013-C support
Fixes: #22