syssi/esphome-yeelight-ceiling-light #23: Support for Jiaoyue 650

PsychoRS, 2021-10-04:
Hi,

Anyone got Jiaoyue 650 (white + RGB backlight), yeelink.light.ceiling4 work with ESPHome?

molnarti, 2021-10-04:
i have those lights, but i think you need to replace the ESP32 board inside the lamp, as it only has a single core ESP32 cpu that does not work (or does not work well) with ESPhome

syssi, 2021-10-04:
All yeelight devices (with esp32 mcu) are single core versions. For me they are pretty stable if you don't use the bluetooth stack.

PsychoRS, 2021-10-04:
So can you confirm that this light comes with an ESP32 and not a mediatek controller?

syssi, 2021-11-03:
If you don't want to open your device you could use python-miio to request the device info (`miiocli info`). In best case the `hw_ver` gives some info about the used microcontroller.

TheZoker, 2023-04-25:
Hi @syssi 
I have this light and this is the output of the `info` call:
```
Model: yeelink.light.ceiling4
Hardware version: MTK7697
Firmware version: 1.6.6_0056
```

Does this help? Is there any chance that this light will work with ESPHome?

PsychoRS, 2023-04-25:
> Hi @syssi I have this light and this is the output of the `info` call:
> 
> ```
> Model: yeelink.light.ceiling4
> Hardware version: MTK7697
> Firmware version: 1.6.6_0056
> ```
> 
> Does this help? Is there any chance that this light will work with ESPHome?

Its a Mediatek SoC, so no chance of running ESPHome unless you replace the Mediatek board with an ESP32 one.