syssi/esphome-yeelight-ceiling-light #28: webserver component not valid on Yeelight Meteorite?

PsychoRS, 2021-11-09:
Hi,

I'm trying to compile ESPHome firmware for my Meteorite. I'm unable to do it using ESP-IDF framework if I keep the webserver: component.

Is it normal? Does the new ESP-IDF platform support webserver component?

Thanks

syssi, 2021-11-09:
If a component isn't supported by ESP-IDF you should see a critical error telling the platform "arduino" is supported only.

PsychoRS, 2021-11-09:
Yes, that's the error I got, just asking if it is normal behaviour.

syssi, 2021-11-09:
Yes it is. You could use the arduino platform if you want to avoid ESP-IDF:

https://github.com/syssi/esphome-yeelight-ceiling-light/commit/db03752d2354cd0a5aa51fbc723b2520a3e677b0

syssi, 2021-11-09:
Do you agree to close the issue?

PsychoRS, 2021-11-09:
Thanks, sorry.

But... the lamp is 100% operative using the older arduino platform instead of the new ESP-IDF?

syssi, 2021-11-09:
I've asked more or less the same question here: https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/62

closed 2021-11-09 by syssi