syssi/esphome-yeelight-ceiling-light #29: Help for YLXD76YL rx tx

gudszent, 2021-12-19:
Hi All, 

Somebody have a picture for YLXD76YL to clear which is the rx tx and vcc pins?
From this [picture](https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/ylxd76yl/010.jpg) and sheet not clear for me, what should I do :)
I think something like [that](https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/ylxd41yl/003.jpg)

Thank you for your time

O.

syssi, 2021-12-19:
This image will help a bit in the meantime:

https://raw.githubusercontent.com/AchimPieters/esp32-homekit-camera/master/Images/ESP32-VROOM-32D-PINOUT.png

Do you see `GPIO1` and `GPIO3`? This is TX and RX (wired directly to the ESP32).

syssi, 2021-12-19:
The pin header is labeled at the bottom of the PCB:

[image]

To make it explicit: The yellow wire should be GND. You can verify it be measuring the resistance / continuity between the pin header pin and the `GND` pin (`pin 1`) of the ESP32. Please to the same for `3V3` to be 100% sure.

syssi, 2021-12-19:
Could I provide anything else?

gudszent, 2021-12-19:
> Could I provide anything else?

Thank you, I guess I can start with it :) 
The next step the custom firmware build, as I see, no prebuild img.

Thank you for your time and hard work

O.

syssi, 2021-12-19:
You will learn to handle esphome pretty fast and you will see building and flashing a custom image requires a single command call.

gudszent, 2021-12-19:
> You will learn to handle esphome pretty fast and you will see building and flashing a custom image requires a single command call.

you are right, I have some docker stuff, so that was easy. 

Now come the bootloader part, I try to find TP3 label, because the gpio direct connect to grn on esp not work for me :/

syssi, 2021-12-19:
TP3 is on the back of the daughter board where the ESP32 sits on.

closed 2021-12-20 by gudszent

syssi, 2021-12-20:
I hope you were successful. :-)

gudszent, 2021-12-20:
> I hope you were successful. :-)

Sure, thank you for your help, I hope you will enjoy your coffee ;)

now I have to modify it, because I use openHAB, so I need mqtt and esp-idf not support is :)

syssi, 2021-12-20:
Thanks for your support! I will enjoy the coffee for sure! Wait a second I will provide some config lines to change the platform from ESP-IDF to Arduino. The MQTT component is available there.

syssi, 2021-12-20:
https://github.com/syssi/esphome-yeelight-ceiling-light/commit/db03752d2354cd0a5aa51fbc723b2520a3e677b0#diff-b062fda38115034a80dc109f77ecaab2fba849bd59695a5d7db5d4c49883a636

Just replace

```
# esp-idf framework
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```

with

```
# arduino framework
esp32:
  board: esp32doit-devkit-v1
  framework:
    # https://github.com/arendst/Tasmota/blob/development/platformio_tasmota32.ini:
    # core32.platform = espressif32 @ 3.3.1
    type: arduino
    version: 3.3.1
    # https://github.com/arendst/Tasmota/blob/development/platformio_tasmota_env32.ini:
    # env:tasmota32solo1.platform_packages = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/1.0.7.5/tasmota-arduinoespressif32-solo1-release_v3.3.5.tar.gz
    source: https://github.com/tasmota/arduino-esp32/releases/download/1.0.7.5/tasmota-arduinoespressif32-solo1-release_v3.3.5.tar.gz
```

Now it should be possible to use the mqtt component:

```
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
```

syssi, 2021-12-20:
Warning: Please read the docs of the `mqtt` and `api` component. You should remove the `api` component to avoid reboots because of missing client connections.

gudszent, 2021-12-20:
> Warning: Please read the docs of the `mqtt` and `api` component. You should remove the `api` component to avoid reboots because of missing client connections.

I avoided it different way, I hope it will works, I like the ota/api update :P
api:
  reboot_timeout: 0s
I had issue with hidden wifi, what default not "on"
wifi:
  ssid: wifiname
  password: wifipass
  fast_connect: on 

I stole the arduino part from your previous commit, but thanks!

Now the mqtt work and I can control the light with command, but in json format, so now read the documentation, how can I change it