syssi/esphome-yeelight-ceiling-light #30: Support for Yeelight Smart LED Light Panels

Apocrathia, 2021-12-22:
The new Yeelight Smart LED Light Panels (Model YLFWD-012) appear to have been released with the non-local firmware on the controller. However, upon further inspection, the controller is running an ESP-WROOM-32D. What can we do to get this working with ESPhome?

[Please refer to the following post on the ESPhome forums for more details.](https://community.home-assistant.io/t/yeelight-smart-led-light-panels/370049)

syssi, 2021-12-22:
I don't own the device so I would like to guide you through the complete process to transform your lamp to an esphome device. The most important and required tool is a multimeter. Do you own a multimeter and do you know how to measure voltage & resistance? :-)

syssi, 2021-12-22:
1. Please provide a high resolution photo of the PCB holding the ESP32 (both sides)
2. We will try to identify the following pins: `3.3V`, `GND`, `RXD`, `TXD`, `GPIO0` + the GPIOs driving the LEDs

- `3.3V` & `GND` is required to power the board while flashing. If the device has a power supply and is powered by a low voltage we could use the power supply to power the board/esp32.
- `GPIO0` is required to boot the ESP32 into the `download mode` to flash new firmware / make a backup of the stock firmware.
- The GPIOs driving the LEDs: Could you describe the features of the stock firmware? Do you know any details about the LEDs? Could you provide a photo of the LEDs + the connector to attach the LEDs?

Apocrathia, 2021-12-24:
I've already got all of the necessary tools. Here are all of the pictures that I've taken of the board and accessories.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Additionally, here's a reference for the pinout for the ESP-WROOM-32D
[image]
It looks like the tx/rx pins are already identified on the board, but I might have to solder leads directly to the ESP for the others.
As of right now, it looks like I am able to get them into HA via the Yeelight integration. It took some digging in the app, but it seems that local control is still available to be enabled. However, I also assume that there will be others purchasing these same lights, and wanting them to work with ESPhome.

imhotep, 2022-01-21:
@Apocrathia Did you ever get it to work with ESPHome?

Apocrathia, 2022-01-21:
No, sorry, I haven't really had a chance to work with these. I still haven't even put them up anywhere. I was able to get them pulled into Home Assistant via the Yeelight integration, but I can only control static colors, brightness, and power. None of the effects or audio functions. Because of the number of features provided by the controller, [I've also opened an issue on the python-yeelight repository](https://gitlab.com/stavros/python-yeelight/-/issues/70), as rimplementing everything in ESPHome may be a much larger endeavor than initially expected. However, I simply haven't had an opportunity to really look at it.

k37chup, 2022-11-30:
Same boat hoping that maybe there was some integration esphome or tasmota but haven't seen anything. I was able to get my hands on a 2nd control board because the first one that was sent to me wasn't broadcasting SSID and I think looking at the chip maybe its due to some overlapping solder joints on the ESP32. Bought some replacements and was hoping to flash something other than the Stock Firmware on there. Anyone have the opportunity to look at these yet?

syssi, 2026-11-01:
There is a project that answers most of the open questions here: https://github.com/acidmiku/yeelight-panels-wled

Short summary for the YLFWD-012 (`yeelink.light.plate2`):

- The ESP32 doesn't drive the panels. It's only the Wi-Fi/cloud gateway and talks to a separate microcontroller in the controller, which reads the four buttons and renders the colors and effects.
- The link between them is a small binary protocol on UART1: 9600 baud 8N1, ESP32 TX = GPIO17, RX = GPIO16. The `T`/`R` test pads next to the module tap these lines. Frames look like `55 AA <cmd> <len> <data> <checksum>`. The ESP32 sends power, brightness, color temperature, RGB, effect number and music mode; the MCU reports button presses and state changes back. Full command tables and example frames: https://github.com/acidmiku/yeelight-panels-wled/blob/main/docs/yeelight/PROTOCOL.md
- Every command applies to the whole chain. The protocol has no per-panel addressing, and effects and music modes run inside the MCU, selected by number only. So a firmware that just talks to the MCU (e.g. an ESPHome UART component) can only control all panels at once: one color, one brightness, one built-in effect.
- The panels themselves are simple DMX512 decoders (chip marked `DMX512PS`, 3 channels per panel, addresses assigned automatically along the chain). The project above lifts one resistor to disconnect the MCU from the panel data line and lets the ESP32 output DMX512 on GPIO19 instead. With that mod every panel is its own pixel (WLED in their case, but the same works for any firmware that can output DMX512). The MCU keeps reporting button presses over the UART afterwards.

Without the hardware mod: all panels at once via the UART protocol. With the mod: per-panel control via DMX512. Their guide has the hardware details (flash access, eFuse notes, single-core ESP32, the resistor location): https://github.com/acidmiku/yeelight-panels-wled/blob/main/docs/yeelight/GUIDE.md