syssi/esphome-yeelight-ceiling-light #37: Crashes with esp32_ble_tracker

jvitkauskas, 2022-01-30:
Seeing a lot of crashes with proposed esp32_ble_tracker configuration on esp32-wroom-32d. Most of the time I can't even finish OTA update, I get a crash and `connection reset by peer` message. Tried using default esp32_ble_tracker configuration by just leaving ```esp32_ble_tracker:``` in the yaml and I get no crashes, but lots of remote button presses are missed.

Is esp32 just not up to the task?

syssi, 2022-01-30:
Enabling/using `esp32_ble_tracker` on a single core ESP32 isn't suggested. Using a dual core ESP32 works fine for me.

jvitkauskas, 2022-01-30:
Yes, isn't esp32-wroom-32d dual core?

syssi, 2022-01-30:
All yeelight devices are equipped with single core ESP32. So I will rephrase my question: Do you use the ESP32 of your yeelight or do you use a dedicated board/esp32?

syssi, 2022-01-30:
Please use parts (`esp32_ble_tracker.scan_parameters`) of this configuration as starting point:

https://github.com/syssi/esphome-yeelight-ceiling-light/pull/32/files#diff-28732135f1ce36041cea23f051289d9b9c27d5c84cac31f1e15cfecd068f15f4

syssi, 2022-01-30:
It looks like the

```
  framework:
    type: esp-idf
```

framework is broken / doesn't compile at the most recent esphome version. I hope this will be fixed soon.

jvitkauskas, 2022-01-30:
> All yeelight devices are equipped with single core ESP32. So I will rephrase my question: Do you use the ESP32 of your yeelight or do you use a dedicated board/esp32?

I have bought a separate esp32 module and replaced it with that one.

syssi, 2022-01-30:
Good job! Could you provide your configuration? I would like to try to reproduce the issue.

jvitkauskas, 2022-01-30:
give me a few minutes

jvitkauskas, 2022-01-30:
here's mine with your recommended ble_tracker values:
https://gist.github.com/jvitkauskas/95d7f8952f5ac69f926663039839b320

It's C2001C450 + ylyk01yl + ylkg07yl, but reproducable with C2001C450 + ylyk01yl too.

jvitkauskas, 2022-01-30:
to reproduce the issue just install that yaml and keep pressing buttons on a remote for a minute or so and you'll see a connection loss
```INFO esphome-web-0aad1c.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esphome-web-0aad1c.local
WARNING Disconnected from API
INFO esphome-web-0aad1c.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esphome-web-0aad1c.local```

jvitkauskas, 2022-02-05:
@syssi could you reproduce the issue? I am not sure, maybe it's related to the chip I am using. I see there are newer ones available.

syssi, 2022-02-06:
I had to reduce you config a bit because I've flashed the configuration to a wemos d1 mini. The reduced config is stable:

```
$ esphome logs test-ylyk01yl.yaml    
INFO Reading configuration test-ylyk01yl.yaml...
INFO Updating https://github.com/syssi/esphome-yeelight-ceiling-light.git@add-yeedimmer-ylkg07yl-support
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (CP2104 USB to UART Bridge Controller - CP2104 USB to UART Bridge Controller)
  [2] Over The Air (test-ylyk01yl.local)
(number): 2
INFO Starting log output from test-ylyk01yl.local using esphome API
INFO Successfully connected to test-ylyk01yl.local
[12:45:02][I][app:102]: ESPHome version 2022.1.1 compiled on Feb  6 2022, 12:43:43
[12:45:02][C][wifi:488]: WiFi:
[12:45:02][C][wifi:350]:   Local MAC: FF:FF:FF:FF:FF:FF
[12:45:02][C][wifi:351]:   SSID: '...'
[12:45:02][C][wifi:352]:   IP Address: ...
[12:45:02][C][wifi:354]:   BSSID: FF:FF:FF:FF:FF:FF
[12:45:02][C][wifi:355]:   Hostname: 'test-ylyk01yl'
[12:45:02][C][wifi:357]:   Signal strength: -68 dB ▂▄▆█
[12:45:02][C][wifi:361]:   Channel: 3
[12:45:02][C][wifi:362]:   Subnet: 255.255.255.0
[12:45:02][C][wifi:363]:   Gateway: ...
[12:45:02][C][wifi:364]:   DNS1: ...
[12:45:02][C][wifi:365]:   DNS2: 0.0.0.0
[12:45:02][C][logger:236]:   Hardware UART: UART0
[12:45:02][C][xiaomi_ylyk01yl:012]: Xiaomi YLYK01YL
[12:45:02][C][xiaomi_ylyk01yl:013]:   Keycode 'last button pressed'
[12:45:02][C][xiaomi_ylyk01yl:013]:     State Class: ''
[12:45:02][C][xiaomi_ylyk01yl:013]:     Unit of Measurement: ''
[12:45:02][C][xiaomi_ylyk01yl:013]:     Accuracy Decimals: 1
[12:45:02][C][esp32_ble_tracker:706]: BLE Tracker:
[12:45:02][C][esp32_ble_tracker:707]:   Scan Duration: 60 s
[12:45:02][C][esp32_ble_tracker:708]:   Scan Interval: 250.0 ms
[12:45:02][C][esp32_ble_tracker:709]:   Scan Window: 250.0 ms
[12:45:02][C][esp32_ble_tracker:710]:   Scan Type: PASSIVE
[12:45:02][C][ota:082]: Over-The-Air Updates:
[12:45:02][C][ota:083]:   Address: test-ylyk01yl.local:3232
[12:45:02][C][api:134]: API Server:
[12:45:02][C][api:135]:   Address: test-ylyk01yl.local:6053
[12:45:02][C][api:139]:   Using noise encryption: NO
[12:45:02][C][mdns:084]: mDNS:
[12:45:02][C][mdns:085]:   Hostname: test-ylyk01yl
[12:45:04][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:04][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:04][D][main:275]: Button minus pressed
[12:45:04][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:04][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:04][D][main:275]: Button minus pressed
[12:45:05][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:05][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:05][D][main:275]: Button minus pressed
[12:45:05][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:05][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:05][D][main:275]: Button minus pressed
[12:45:05][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:05][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:05][D][main:275]: Button minus pressed
[12:45:05][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:05][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:05][D][main:275]: Button minus pressed
[12:45:06][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:06][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:06][D][main:275]: Button minus pressed
[12:45:06][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:06][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:06][D][main:275]: Button minus pressed
[12:45:06][D][xiaomi_ble:324]: Got Xiaomi YLYK01YL (64:88:FF:6A:CB:A5):
[12:45:06][D][sensor:125]: 'last button pressed': Sending state 5.00000  with 1 decimals of accuracy
[12:45:06][D][main:275]: Button minus pressed
[... 195 lines cut ...]
    refresh: 0s

xiaomi_ylyk01yl:
  mac_address: "${remote_mac}"
  last_button_pressed:
    name: "last button pressed"
  on_button_on:
    then:
      - logger.log: "Button on pressed"
  on_button_off:
    then:
      - logger.log: "Button off pressed"
  on_button_sun:
    then:
      - logger.log: "Button sun pressed"
  on_button_m:
    then:
      - logger.log: "Button m pressed"
  on_button_plus:
    then:
      - logger.log: "Button plus pressed"
  on_button_minus:
    then:
      - logger.log: "Button minus pressed"
```

closed 2022-05-23 by syssi

jvitkauskas, 2022-06-03:
Replaced chip with ESP32-S and all issues disappeared