syssi/esphome-yeelight-ceiling-light #40: Error 'platformio/toolchain-esp32ulp @ ~1.22851.0' on 'linux_aarch64'

Weasy666, 2022-05-30:
I am trying to build `yeelight_light_ceiling10.yaml` on my raspberry pi3 with the ESPHome Homeassistant addon, but am getting the following error:

```bash
INFO Reading configuration /config/esphome/yeelight-crystal-pendent.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing yeelight-light-ceiling10 (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

## My steps
I want to build the binary on the raspberry and then flash it to the yeelight device per ESPHome Web. So i open the ESPHome Addon, click `New Device` > `Install` > `Manual Download` > `Modern Format` and get the above error.  
The Raspberry is running Home Assistant OS and everything is up-to-date. I am pretty new to the ESPHome stuff, so i hope it is ok to ask for help here.

syssi, 2022-05-30:
The toolchain isn't Raspberry Pi 3 (aarch64) compatible at the moment. Unfortunately you will have to use another machine to compile & flash the yaml. I try to provide a list of steps to pave your way:

1. Pick another machine (Linux if possible, Windows if there is no other way)
2. Install python3
3. Install ESPHome using the python package manage: `pip3 install esphome`
4. Execute `esphome` at a terminal and build + flash the yaml: `esphome run yeelight_light_ceiling10.yaml`

Weasy666, 2022-05-30:
I build it with WSL2 with `esphome compile yeelight_light_ceiling10.yaml` and my `secret.yaml`. Because WSL did not recognize my CP2102N device, i had to flash the `./.esphome/build/yeelight-light-ceiling10/.pioenvs/yeelight-light-ceiling10/firmware.bin` with ESPHome Web. I put the board back into the lamp and connected the power cord, but the lamp does not light up or appear in my WiFi, as a Hotspot or as a Bluetooth device.  
Is there something else i am missing?

syssi, 2022-05-30:
I sounds like the ESP32 is unable to connect to your WiFi. You should attach the USB-to-serial adapter again to look into the logs (`esphome logs yeelight_light_ceiling10.yaml`).

Weasy666, 2022-05-30:
Logs are spammed with

```
[15:05:36]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:05:36]flash read err, 1000
[15:05:36]ets_main.c 371
```

syssi, 2022-05-30:
Please flash the firmware again. Something went wrong last time. If this doesn't help try `erase_flash` via `esptool.py`.

Weasy666, 2022-05-30:
🤦‍♂️ i am an idiot! you need to flash `firmware-factory.bin` and NOT `firmware.bin`!  

Sorry for that! And thank you very much for your help 🙂

syssi, 2022-05-30:
You are welcome!