syssi/esphome-yeelight-ceiling-light #42: please support yeelink.light.fancl5

0neday, 2022-07-18:
RT,

thank you

syssi, 2022-07-18:
You could open the device and provide some photos. If you own a multimeter I could provide some instructions how to collect all required infos.

0neday, 2022-07-18:
> You could open the device and provide some photos. If you own a multimeter I could provide some instructions how to collect all required infos.

thank you, I have multimeter, I will give you  some postion of gpio, such as gpio0,  rx, tx ...

syssi, 2022-07-19:
Some high-res photos of the mainboard especially the WiFi daughter board are important. In best case the pin header between the mainboard and the daughter board is labeled.

Are you able to control different speed levels of the ceiling fan using the app?

0neday, 2022-07-19:
> Are you able to control different speed levels of the ceiling fan using the app?

using Mijia app, I can control fan speed.

syssi, 2022-07-19:
How many levels?

0neday, 2022-07-19:
循环风三档，标准风 0-100%调速。

拆了看了下，esp32通过 软 串口给风扇驱动模块。目前只分析出，白线是接到 IC2 的8号引脚，其他分析太难了，没搞过。
图片已经上传。
https://github.com/0neday/yeelight-c900

syssi, 2022-07-19:
You could use a logic analyzer (something like this https://de.aliexpress.com/item/1005002511491492.html or https://de.aliexpress.com/item/1005003812410259.html) plus a open-source software called sigrok/pulseview to capture the serial traffic between the ESP and the driver board of the fan. As soon as we know the messages here we can try to replicate the communication. The ceiling lamp is working already? What's working and what's not if you flash your yaml? Did you make a backup of the stock firmware?

0neday, 2022-07-19:
no, I have not flash esphome, I just test UART using a usb-ttl tools, and get 
```
brownout detector 

```
when I set gpio0 = low, and it  could enter into flash mode.

I can not get gpio is for light.  the gpio of my yaml is copy from your's, do not test.

Do you any decompiled circuit diagram of yeelight ceiling?

syssi, 2022-07-19:
Another wish/suggestion:

Please use a continuity tester or a ohm meter to measure the continuity between the test points (`TP1`, `TP5`) and the GPIOs of the ESP. I bet `GPIO0` is exposed as test point.

0neday, 2022-07-20:
> Measure the voltage between GND and every GPIO. You should find a GPIO providing 3.3V (PWM=100%).

you are genius！

0neday, 2022-07-20:
```
[16:05:37]ets 016 :22:MS
[16:05:37]. (POWQ}IMET),0L&BA%_FASTe1M!}BOOT)C˥sip:, SPR
[16:05:37]clE0,qߖ0,d_d.0,cs0_0x,hd_0x,wp_0xj
[16:05:37]moY'T	div:2C!+0x3ffbbn:4!+0x3fffc,lKj
[16:05:37]ho 12 roo[j
[16:05:37]l+0x,lenj
[16:05:37]load:008bb:587C 0x4006d4Cfast_reboot_count in psm is now 1
[16:05:37]08:00:00.100 [I] miio_ot: httpdns enabled
[16:05:37]08:00:00.100 [I] miio_ot: dlg enabled
[16:05:37]
[16:05:37]
[16:05:37]_|      _|  _|_|_|  _|_|_|    _|_|  
[16:05:37]_|_|  _|_|    _|      _|    _|    _|
[16:05:37]_|  _|  _|    _|      _|    _|    _|
[16:05:37]_|      _|    _|      _|    _|    _|
[16:05:37]_|      _|  _|_|_|  _|_|_|    _|_|  
[16:05:37]JENKINS BUILD NUMBER: N/A
[16:05:37]BUILD TIME: Sep 24 2020,16:03:38
[16:05:37]BUILT BY: N/A
[16:05:37]MIIO APP VER: 2.0.6_0006
[16:05:37]MIIO MCU VER: 
[16:05:37]MIIO DID: 474769981
[16:05:37]MIIO WIFI MAC: ec4d3e7e1033
[16:05:37]MIIO MODEL: yeelink.light.fancl5
[16:05:37]ARCH TYPE: esp32,0x0000a601
[16:05:37]ARCH VER: 401ef84
[16:05:37]W (637) phy_init: saving new calibration data because of checksum failure, mode(0)
[16:05:38]W (957) wifi: ap start fail
[16:05:38]
[16:05:38]W (957) wifi: ap start fail
[16:05:38]
[16:05:38]08:00:00.650 [I] otu: Opened.
[16:05:42]08:00:05.040 [W] arch_flash: data from DROM. (arch_flash_write,74)
```
read from usb-ttl

0neday, 2022-07-20:
test rulst is :
```
cold --> gpio21
warn --> gpio19
night mode ---> gpio23
```
so, the light gpio is same as `yeelink.light.ceiling15`

syssi, 2022-07-20:
Could you provide a photo of your remote control? I assume it's the standard BLE remote talking to the ESP. It's known how to receive commands from these kind of remotes but we shouldn't try to support the remote control at the first iteration. I would be happy if we focus on the ceiling light and fan feature.

syssi, 2022-07-20:
Did you backup the stock firmware?

syssi, 2022-07-20:
It looks like you managed to flash the ESP32. If you don't want to buy a logic analyzer I could provide some instruction how to piggy back another ESP (running esphome to dump some traffic) to the serial line between the ESP and the fan driver board. The idea is to sniff the traffic between these two components to get a better understanding how to replicate the communication.

0neday, 2022-07-20:
> back another ESP (running esphome to dump some traffic) to the serial line between the ESP

that is a good idea! I have a nodemcu esp32 , I will to flash stock firmware into it , and check what happed

0neday, 2022-07-20:
the buzzer beep is for fan mcu BP6601, check here https://bbs.elecfans.com/jishu_1868688_1_1.html

0neday, 2022-07-20:
> you want to receive commands using Home Assistant you could use this custom component
 no, I want to use esphome to execute animation.

I use ble tracker to get remote controller adversie data, like 
```
# for fan
[21:53:04][D][ble_adv:107]:     - 0xFE95: (length 17)  - 50.30.8E.06.8B.0B.85.6E.38.C1.A4.01.10.03.00.00.00 (17)
# for light
[21:52:09][D][ble_adv:107]:     - 0xFE95: (length 17)  - 50.30.8E.06.89.0B.85.6E.38.C1.A4.01.10.03.01.00.00 (17)
# for recirculating air
[21:54:25][D][ble_adv:107]:     - 0xFE95: (length 17)  - 50.30.8E.06.8E.0B.85.6E.38.C1.A4.01.10.03.02.00.00 (17)
....

```
so, just use esphome to get adversie data and using `xor ` get  third-to-last byte and compare, and using on_... to execute .

syssi, 2022-07-20:
In the past the single core ESP32 of the yeelight devices did crash if you try to use the BLE tracker at your configuration yaml. Some remote controls are encrypted and the `bind_key` is required to decrypt the data. Do you know my implementation of the ceiling lamp remote control?

syssi, 2022-07-20:
> I use framework is - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6 that support ota web.

I prefer ESP-IDF over the Arduino framework. I flash my esphome nodes using `esphome run config.yaml` + `ota:` component.

0neday, 2022-07-20:
can not downlod, but you could look this online https://www.docin.com/p-2267808805.html

0neday, 2022-07-20:
maybe, we cloud us decompile engineering to check how control fan 
https://github.com/jsandin/esp-bin2elf

syssi, 2022-07-20:
I've no experience with reverse engineering of ELF executables. This is the way I would choose:

1. Identify the GPIOs attached to `con2` the pin header between the mainboard and the fan driver board.
2. Flash the backup of the stock firmware to another ESP32. Try to boot. It's likely the ESP doesn't boot/crashs because peripherals (EEPROM?) isn't available or the MAC address doesn't match. If the ESP boots connect the USB-TTL converter to the RX/TX pins of the `con2` GPIOs. Use the android app to start/stop/control the fan. Try different baudrates. Do you see some messages/control instructions?
3. If step 2 isn't successful flash the stock firmware back to the yeelight single core ESP32. Let the ESP boot and sniff the traffic at `con2` again without having the fan driver board attached.

Some more ideas:

4. Attach dhe fan driver board using con2 to the ESP32 again and power the board with 5V. Don't apply the 24V. I hope this doesn't harm the device but I don't know for sure. You could attach the USB-TTL converter to the RX/TX lines of `con2`. Alternatively you could use another ESP (f.e. ESP8266) plus a configuration yaml like this

https://github.com/syssi/esphome-jbd-bms/blob/main/tests/esp8266-dummy-receiver.yaml

to sniff the traffic of one direction. Try to identify the baudrate. Capture the traffic/commands from the ESP. Swap the RX/TX lines and capture the responses from the fan driver board (hopefully the board responds). If this doesn't work I would attach a 24V power source too to make the fan driver board happy.

0neday, 2022-07-20:
> Identify the GPIOs attached to con2 the pin header

I try to identify, but failed! the board have a layer of insulating glue.

> You could attach the USB-TTL converter to the RX/TX lines of con2.

 bwteen rx, tx of `con2 ` and gpio, there is some divider resistor, the voltage rx and tx is `2V`

0neday, 2022-07-20:
> Do you know my implementation of the ceiling lamp remote control? 

I have check your external_components, but I have no idea how to get it work.

I use 
```
# BLE    
esp32_ble_tracker:
  on_ble_advertise:
    - mac_address: A4:C1:38:6E:85:0B
      then:
        - lambda: |-
            for (auto data : x.get_service_datas()) {
                ESP_LOGD("ble_adv", "    - %s: (length %i)  - %s", data.uuid.to_string().c_str(), data.data.size(),hexencode(data.data).c_str());
            }
```
to print adv data.

for `50.30.8E.06.89.0B.85.6E.38.C1.A4.01.10.03.01.00.00` of fan key data,  how to use `this->keycode_->publish_state(*res->keycode)` to get keycode, and how create on_... events? how to modify your code, thank you!

0neday, 2022-07-21:
> > the buzzer beep is for fan mcu BP6601, check here https://bbs.elecfans.com/jishu_1868688_1_1.html
> 
> Could you ~help me to download~ provide the two attachments?

https://github.com/0neday/yeelight-c900/commit/d381f730cc247fe4ff168e5fe835aaec6f1c73ce

syssi, 2022-07-21:
> bewteen rx, tx of con2 and gpio, there is some divider resistor, the voltage rx and tx is 2V

I assume the fan driver board runs at 5V and the logic level of the serial interface of the BP6601 is 5V too. To shift the logic level to 3.3V (because the ESP isn't 5V tolerant) the voltage divider is used. You could measure the voltage between TX and GND at the BP6601 to prove this assumption.

syssi, 2022-07-21:
> when I tap one time, but on_ble_service_data_advertise advertise 2 times data, so when I use light.toggle, that work poorly

There is a frame counter which increases permanently:

https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/components/xiaomi_ble/xiaomi_ble.cpp#L154

The external component tracks the counter value and discards duplicates. I'm busy today and will provide some more input how to improve and use the external component tomorrow.

syssi, 2022-07-21:
How to use / extend the external component:

This instructions clones the remote repository every time you compile your configuration:

```
external_components:
  - source: github://syssi/esphome-yeelight-ceiling-light@main
    refresh: 0s
```

To be able to make local changes you have to clone the repository manually:

```
git clone git@github.com:syssi/esphome-yeelight-ceiling-light.git
```

And let the external component configuration point to the local checkout / directory:

```
external_components:
  - source: components
    refresh: 0s
```

Now you can make changes here:

```
components/xiaomi_ble/xiaomi_ble.cpp
```

We need a new block for your device likes this:

```
  } else if ((raw[2] == 0x53) && (raw[3] == 0x01)) {  // Yeelight Remote Control YLYK01YL
    result.type = XiaomiParseResult::TYPE_YLYK01YL;
    result.name = "YLYK01YL";
```

Your device type idenfier is `0x068E`. As next step (I assume) we need a new directory for the device implementation. May be your remote control is pretty similar to the YLYK01YL. In this case we could just extend the existing implementation. This diff should help to get a better understand what a full device implementation looks like:

https://github.com/esphome/esphome/pull/1783/files#diff-a1fc96aecdd718ce2fd8d4892c7d9ecef75683f467cc58282fd4c0b13cd24934

syssi, 2022-07-21:
Could you provide the bluetooth name of your remote control? I assume it's advertised and part of your ESPHome log at the very beginning.

syssi, 2022-07-21:
I assume this GPIO must be enable/pulled to start/enable the fan driver board:

```
EN( pin3 )  ----> con2( wifi_en )
```

0neday, 2022-07-21:
> Could you provide the bluetooth name of your remote control? I assume it's advertised and part of your ESPHome log at the very beginning.

yee-rc

0neday, 2022-07-21:
> The external component tracks the counter value and discards duplicates. I'm busy today and will provide some more input how to improve and use the external component tomorrow. 

have a good day, thank you very much

syssi, 2022-07-21:
> yee-rc

Is there a model identifier on the back (under the sticker?) or somewhere?

0neday, 2022-07-21:
> > when I tap one time, but on_ble_service_data_advertise advertise 2 times data, so when I use light.toggle, that work poorly
> 
> There is a counter at the frame which is increased per button press:
> 
> https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/components/xiaomi_ble/xiaomi_ble.cpp#L154
> 
> The external component tracks the counter value and discards duplicates. I'm busy today and will provide some more input how to improve and use the external component tomorrow.

thank you, I have fixed by defined global variable `frame_counter`
and use `if condition and ` 

```
globals:
   - id: frame_counter
     type: int
     initial_value: '0'

# BLE    
esp32_ble_tracker:
  on_ble_service_data_advertise:
    - mac_address: A4:C1:38:6E:85:0B
      service_uuid: FE95
      then:
         - if:
            condition: 
              and:
                - lambda: "return(id(frame_counter) != x[4]);"
                - lambda: "return(x[14] == 1);"
            then:
             - lambda: "id(frame_counter) = x[4];"
             - switch.turn_on: buzzer_beep
             - lambda: 'id(yeelight_remote_controller).publish_state(x[14]);'
             - light.toggle: ceiling_light
                      
                 
         - if:
            condition: 
              and:
                - lambda: "return(id(frame_counter) != x[4]);"
                - lambda: "return(x[14] == 4);"
            then:
             - lambda: "id(frame_counter) = x[4];"
             - switch.turn_on: buzzer_beep
             - lambda: 'id(yeelight_remote_controller).publish_state(x[14]);'
             - light.toggle: night_light
            
```

0neday, 2022-07-21:
> > yee-rc
> 
> Is there a model identifier on the back (under the sticker?) or somewhere?

no any identifier,  maybe is a DIY product, just know the chip of ble is `TLSR8267`

0neday, 2022-07-21:
> back another ESP (running esphome to dump some traffic) to the serial line between the ESP

maybe not work on nodemcu esp32

```
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
```

0neday, 2022-07-21:
I found this issue, that show uart communication is 4 bytes .https://github.com/arendst/Tasmota/discussions/13664#discussioncomment-2929489

do you know how to install stock firmware, thank you

0neday, 2022-07-21:
sniff  uart communication data fromat is 
```
[D][uart_debug:114]: <<< 01:01:01:13:11:03 关
[D][uart_debug:114]: <<< 01:03:01:68:64:03 开

自然风 三档
[D][uart_debug:114]: <<< 01:03:01:05:01:03
[D][uart_debug:114]: <<< 01:03:01:36:32:03
[D][uart_debug:114]: <<< 01:03:01:68:64:03
```

syssi, 2022-07-21:
You can send these frames using `uart.write`: See https://github.com/syssi/esphome-jbd-bms/blob/main/tests/esp8266-query-data.yaml

0neday, 2022-07-21:
thank you! fan is work.

syssi, 2022-07-21:
Could you try to provide a much details about the frames as possible?

```
Byte  Value  Description
0     0x01   Start of frame (always 0x01)
1     0x03   0x01=Off, 0x03=On (is 0x02 also a possible value?)
2     0x01   ??
3     0x68   ??
4     0x64   Fan speed in percent 0x01 ...0x64 = 1...100%
5     0x03   End of frame (always 0x03)
```

syssi, 2022-07-21:
It's possible byte3 is the checksum: sum of byte1+byte2+byte4.

0neday, 2022-07-21:
> Could you try to provide a much details about the frames as possible?
> 
> ```
> Byte  Value  Description
> 0     0x01   Start of frame (always 0x01)
> 1     0x03   0x01=Off, 0x03=On (is 0x02 also a possible value?)
> 2     0x01   ??
> 3     0x68   Fan speed?
> 4     0x64   Checksum (n - byte1 - byte2 - byte3)
> 5     0x03   End of frame (always 0x03)
> ```
> 
> Does the uC of the fan driver board respond? The `dummy_receiver` should write the answers to the log.

this need more data to recompile frame format.
because I flash stock firmware into a new esp32, that cannot connect Mijia server( maybe server use mac adress check ) , so Mijia App show my device is downline. so that I can not test 0-100% speed.

```
# open
[D][uart_debug:114]: >>> 01:04:01:18:13:03
[D][uart_debug:114]: <<< 01:F3:01:07:13:03

# close
[D][uart_debug:114]: >>> 01:01:01:13:11:03
[D][uart_debug:114]: <<< 01:F3:01:05:11:03

syssi, 2022-07-21:
I still assume there is an eeprom next to the ESP32 containing some secrets required to talk to the Mijia cloud.

syssi, 2022-07-21:
```
# on
[D][uart_debug:114]: >>> 01:04:01:18:13:03 -> 0x04 + 0x01 + 0x13 = 0x18
[D][uart_debug:114]: <<< 01:F3:01:07:13:03 -> 0xF3 + 0x01 + 0x13 = 0x07

# off
[D][uart_debug:114]: >>> 01:01:01:13:11:03 -> 0x01 + 0x01 + 0x11 = 0x13
[D][uart_debug:114]: <<< 01:F3:01:05:11:03 -> 0xF3 + 0x01 + 0x11 = 0x05

# 1 level
[D][uart_debug:114]: >>> 01:03:01:05:01:03 -> 0x03 + 0x01 + 0x01 = 0x05
[D][uart_debug:114]: <<< 01:F3:01:F5:01:03 -> 0xF3 + 0x01 + 0x01 = 0xF5

# 2 level
[D][uart_debug:114]: >>> 01:03:01:36:32:03 -> 0x03 + 0x01 + 0x32 = 0x36
[D][uart_debug:114]: <<< 01:F3:01:26:32:03 -> 0xF3 + 0x01 + 0x32 = 0x26

# 3 level 
[D][uart_debug:114]: >>> 01:03:01:68:64:03 -> 0x03 + 0x01 + 0x64 = 0x68
[D][uart_debug:114]: <<< 01:F3:01:58:64:03 -> 0xF3 + 0x01 + 0x64 = 0x58

Byte  Value  Description
0     0x01   Start of frame (always 0x01)
1     0x03   0x01=Off, 0x03=On, 0x04=OnReverse
2     0x01   ??
3     0x68   Checksum (byte1 + byte2 + byte4)
4     0x64   Fan speed 0x01...0x64 = 1...100%
5     0x03   End of frame (always 0x03)
```

syssi, 2022-07-21:
Could you try to send an command with an invalid checksum and/or invalid speed (0x65)? I assume the first nibble of 0xF3 indicates success. May be byte 2 of the frame is static and always `0x01`.

0neday, 2022-07-21:
```
# 0x04,  with error chechsum, fan not work
[D][uart_debug:114]: >>> 01:04:01:17:13:03
[D][uart_debug:114]: <<< 01:F3:01:94:A0:03

# 0x04, change speed to 0x01 , no return 
[D][uart_debug:114]: >>> 01:04:01:06:01:03

# 0x03, speed = 0x65 is work
[D][uart_debug:114]: >>> 01:03:01:69:65:03
[D][uart_debug:114]: <<< 01:F3:01:59:65:03

# 0x03, speed =  0x20 , work
[D][uart_debug:114]: >>> 01:03:01:24:20:03
[D][uart_debug:114]: <<< 01:F3:01:14:20:03

# 0x03 , with error checksum, fan not change state
[D][uart_debug:114]: >>> 01:03:01:15:10:03
[D][uart_debug:114]: <<< 01:F3:01:94:A0:03

I guess speed is just for `0x03`, so byte 1 is identify of 自然风 or 循环风 ？

`0x04` is for `on`

the `A0` of retrun code say, not correct, or chechsum error.

0neday, 2022-07-22:
I have another question is how to use uart read to get return code for comfirming the write is correct. 
like  use `uart.read` get return, if get byte 4 is `A0`, that show write is error.

syssi, 2022-07-22:
The next step is the implementation of a custom UART component:

https://esphome.io/custom/uart.html

or a full blown external component to control/monitor the fan. I prefer the external component.

syssi, 2022-07-22:
If the fan is off and you send the set speed to 100% (`01:03:01:68:64:03`) command. Does the fan turn on or is the turn on (`01:04:...`) command required first?

syssi, 2022-07-22:
If you don't want to do the deep dive into the external components of ESPHome I can prepare a first draft of the fan component.

syssi, 2022-07-22:
```
# miot specs of the device / stock firmware features

https://miot-spec.org/miot-spec-v2/instance?type=urn:miot-spec-v2:device:light:0000A001:yeelink-fancl5:1

Device 'urn:miot-spec-v2:device:light:0000A001:yeelink-fancl5:1': Light with 5 services

* Service siid 1: (Device Information): 4 props, 0 actions

        ## Properties ##
                siid 1: piid: 1 (Device Manufacturer): (string, unit: None) (acc: ['read'])
                siid 1: piid: 2 (Device Model): (string, unit: None) (acc: ['read'])
                siid 1: piid: 3 (Device Serial Number): (string, unit: None) (acc: ['read'])
                siid 1: piid: 4 (Current Firmware Version): (string, unit: None) (acc: ['read'])

* Service siid 2: (Light): 4 props, 0 actions

        ## Properties ##
                siid 2: piid: 1 (Switch Status): (bool, unit: None) (acc: ['read', 'write', 'notify'])
                siid 2: piid: 3 (Brightness): (uint8, unit: percentage) (acc: ['read', 'write', 'notify'])
                        Range: [1, 100, 1]
                siid 2: piid: 5 (Color Temperature): (uint32, unit: kelvin) (acc: ['read', 'write', 'notify'])
                        Range: [2700, 6500, 1]
                siid 2: piid: 7 (Mode): (uint8, unit: none) (acc: ['read', 'write', 'notify'])
                        {'value': 0, 'description': 'Day'}
                        {'value': 1, 'description': 'Night'}

* Service siid 3: (Ceiling Fan): 5 props, 0 actions

        ## Properties ##
                siid 3: piid: 1 (Switch Status): (bool, unit: none) (acc: ['read', 'write', 'notify'])
                siid 3: piid: 2 (Fan Level): (uint8, unit: none) (acc: ['read', 'write', 'notify'])
                        {'value': 0, 'description': 'Auto'}
                        {'value': 1, 'description': 'Level1'}
                        {'value': 2, 'description': 'Level2'}
                        {'value': 3, 'description': 'Level3'}
                siid 3: piid: 3 (Status): (uint8, unit: None) (acc: ['read', 'notify'])
                        {'value': 1, 'description': 'Idle'}
                        {'value': 2, 'description': 'Busy'}
                siid 3: piid: 4 (Mode): (uint8, unit: none) (acc: ['read', 'write', 'notify'])
                        {'value': 0, 'description': 'Auto'}
                        {'value': 1, 'description': 'Auto'}
                siid 3: piid: 10 (Power Off Delay Time): (uint32, unit: minutes) (acc: ['read', 'write', 'notify'])
                        Range: [0, 720, 1]

* Service siid 4: (yl-light): 7 props, 9 actions

        ## Properties ##
                siid 4: piid: 1 (): (uint8, unit: none) (acc: ['read', 'notify', 'write'])
                        {'value': 1, 'description': '上电开灯'}
                        {'value': 2, 'description': '上电关灯'}
                siid 4: piid: 3 (): (bool, unit: none) (acc: ['read', 'notify', 'write'])
                siid 4: piid: 4 (): (uint32, unit: minutes) (acc: ['read', 'notify', 'write'])
                        Range: [0, 180, 1]
                siid 4: piid: 5 (): (uint32, unit: minutes) (acc: ['read', 'notify'])
                        Range: [0, 180, 1]
                siid 4: piid: 6 (night-time-str): (string, unit: none) (acc: ['read', 'notify', 'write'])
                siid 4: piid: 7 (dl-brightness): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
                siid 4: piid: 8 (nl-brightness): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]

        ## Actions ##
                siid 4: aiid 1 toggle: in: [] -> out: []
                siid 4: aiid 2 brightness-up: in: [] -> out: []
                siid 4: aiid 3 brightness-down: in: [] -> out: []
                siid 4: aiid 4 brightness-cycle: in: [] -> out: []
                siid 4: aiid 5 ct-cycle: in: [] -> out: []
                siid 4: aiid 7 ct-increase: in: [] -> out: []
                siid 4: aiid 8 ct-decrease: in: [] -> out: []
                siid 4: aiid 9 on-or-bright-cycle: in: [] -> out: []
                siid 4: aiid 10 on-or-ct-cycle: in: [] -> out: []

* Service siid 5: (yl-fan): 5 props, 1 actions

        ## Properties ##
                siid 5: piid: 4 (left-time): (uint32, unit: minutes) (acc: ['read', 'notify'])
                        Range: [0, 720, 1]
                siid 5: piid: 5 (fan-init-power-opt): (uint8, unit: none) (acc: ['read', 'notify', 'write'])
                        {'value': 0, 'description': '上电风扇不开'}
                        {'value': 1, 'description': '上电开风扇'}
                siid 5: piid: 6 (smart-switch-en): (uint8, unit: none) (acc: ['read', 'notify', 'write'])
                        {'value': 0, 'description': '闪断不控制风扇'}
                        {'value': 1, 'description': '闪断控制风扇开关'}
                siid 5: piid: 7 (fan-speed-std): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
                siid 5: piid: 8 (fan-speed-rec): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]

        ## Actions ##
                siid 5: aiid 1 toggle: in: [] -> out: []
```

0neday, 2022-07-22:
> If the fan is off and you send the set speed to 100% (`01:03:01:68:64:03`) command. Does the fan turn on or is the turn on (`01:04:...`) command required first?

need first send `01:04:...`, and then send `01:03:...`, otherwise the fan will reverse

syssi, 2022-07-22:
Could you try to find out how to control the speed if the fan runs in reverse mode? May be byte3 indicates the reverse speeed?

```
Byte  Value  Description
0     0x01   Start of frame (always 0x01)
1     0x03   0x01=Off, 0x03=On, 0x04=OnReverse
2     0x64   Reverse speed 100%
3     0x68   Checksum (byte1 + byte2 + byte4)
4     0x01    Forward speed
5     0x03   End of frame (always 0x03)
```

The miot specs indicates something like: "The fan speed can be controlled in reverse speed too": 

```
                siid 5: piid: 7 (fan-speed-std): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
                siid 5: piid: 8 (fan-speed-rec): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
```

0neday, 2022-07-22:
> external components of ESPHome

how to write a simple  external components of ESPHome

syssi, 2022-07-22:
This is a external component: https://github.com/syssi/esphome-jbd-bms

It can be loaded using:

```
external_components:
  - source: github://syssi/esphome-jbd-bms@main
```

It consumes data from a selected uart:

```
uart:
  id: uart0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jbd_bms:
  uart_id: uart0
```

The folder structure is important: components/<name>/<files>

This must be adapted to support the fancl5 bp6601 frames: https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms/jbd_bms.cpp#L88-L159

This must be adapted to send the fancl frames: https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms/jbd_bms.cpp#L399-L414

syssi, 2022-07-22:
Could you explain the difference between:

```
                siid 3: piid: 2 (Fan Level): (uint8, unit: none) (acc: ['read', 'write', 'notify'])
                        {'value': 0, 'description': 'Auto'}
                        {'value': 1, 'description': 'Level1'}
                        {'value': 2, 'description': 'Level2'}
                        {'value': 3, 'description': 'Level3'}
```

and

```
                siid 5: piid: 7 (fan-speed-std): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
                siid 5: piid: 8 (fan-speed-rec): (uint8, unit: percentage) (acc: ['read', 'notify', 'write'])
                        Range: [1, 100, 1]
```

It looks like the fan has 3 different fan speeds vs. the fan speed can be controlled stepless between 1 and 100%. Do we support all features if we implement the fan speed 1...100% only?

0neday, 2022-07-22:
> It looks like the fan has 3 different fan speeds 
this function is just for BLE remote controller
```
1-level = 1% speed 
2-leve = 50% speed
3-level = 100% speed  
```
that is part of stepless  between 1 and 100%, because remote controller can not do stepless speed control.

> the fan speed can be controlled stepless between 1 and 100%

this function is for Mijia App, in Mijia App you can control fan stepless between 1 and 100% like light brightness controller in HA.

syssi, 2022-07-22:
Please don't risk to brick your device by soldering. We can ignore this missing feature or we could make two attempts to guess the reverse frame. My next bet is passing a inverted value byte4 plus byte2=0x04. Something like this:

```
01:04:01:CRC:F0:03
```

[92 more replies]