syssi/esphome-yeelight-ceiling-light #52: Support for Yeelight Arwen 450/550S

PsychoRS, 2022-07-26:
Hi,

Does anyone know if S variants of Arwen 450/550 are technically the same as the C ones? I've read in some websites, maybe in Yeelight official ones, that they are the same lights with the only difference being thel covers, with a metal ring in the outside of S models.

Can anyone confirm this? Specifications are the same, that is true.

https://us.yeelight.com/shop/arwen-smart-flush-mount-ceiling-light-450s/

Thanks in advance.

helgek, 2022-12-08:
I have both C and S and though I haven’t opened them yet I’m very sure they are the same. The distinction between them is the outer shape. The S models have an extra outer frame which reduces the light emitting angle a bit and the design looks a bit different but apart from that I’m sure they’re the same.

helgek, 2022-12-15:
@syssi: According to miiocli all my four devices are ESP32 based:

Arwen 550S
Model: yeelink.light.ceilb
Hardware version: esp32
Firmware version: 2.1.7_0011

Arwen 550S
Model: yeelink.light.ceilb
Hardware version: esp32
Firmware version: 2.1.7_0011

Arwen 450C
Model: yeelink.light.ceilc
Hardware version: esp32
Firmware version: 2.0.6_0005

Crystal Pendant Light
Model: yeelink.light.ceiling10
Hardware version: ESP32
Firmware version: 1.3.2_0032

syssi, 2022-12-15:
Good job! Thanks for collecting these details. :-)

commit 59e3cb24 (2022-12-16): Add Yeelight Arwen 450S and 550S as supported (Closes: #52)
closed 2022-12-16 by syssi