syssi/esphome-yeelight-ceiling-light #55: Not an issue, just broken pin

rafasanz, 2022-09-15:
Hi!!! nice work!!

With the first ceiling10 was very good flashing with esphome, but not so good with second one. 
The RX pad on back side of PCB of ceiling10 it's broken, so i can't burn firmware on esp32.

GPIO3 should be RX, isn't it? i tried to solder over esp32 but still nothing going.

Some other alternative?

Thanks in advance

syssi, 2022-09-16:
Do you own a multimeter? Please measure the continuity between the TX pad and GPIO1 vs. GPIO3. One of the GPIOs is connected to the TX pad. The other is/was connected to the RX pad. Both GPIOs are located at the upper right corner of the ESP.

syssi, 2022-09-19:
Did you fix your broken light? :-)

rafasanz, 2022-09-19:
yes!! finally i made it today!!
i was knowing why i couldn't upload the code and i figured out that before esp32 GPIO of rx and tx was a resistor so i was able to upload the code by that resistor because i couldn't fix broken RX pad.

thanks!!!

syssi, 2022-09-19:
Good job! Enjoy your **self-managed** lights. :-)

closed 2022-09-19 by syssi