syssi/esphome-yeelight-ceiling-light #56: yeelink.light.ceiling20 vs yeelink.light.ceiling22

B48D81EFCC, 2022-09-20:
Hi,

I have a Xiaomi Mi Ceiling Light.
Originally the device is made by Yeelight. It seems a re-branded (any maybe slightly modified) version of a Yeelight device.

The hostname / DNS of the light is yeelink-light-ceiling22-mibt1ADB (identifies as ceiling22).
On the device itself I can find two slightly different model no.:
- MJXDD001
- MJXDD01SYL

So far everything looks like it is a **yeelink.light.ceiling22**

BUT
According to your published pinout and the photos it fits more the **YLXD50YL**

So some details suggest it is yeelink.light.ceiling20 and some details suggest it is yeelink.light.ceiling22

What do you think?

B48D81EFCC, 2022-09-20:
[image]
[image]
![20220919_17
[image]
5946330_iOS (2)](https://user-images.githubusercontent.com/111175947/191360675-a35c4103-edc4-4770-9f03-2b28864dcf5a.jpg)

syssi, 2022-09-21:
It looks like both versions uses the same WiFi daughter board. The difference between the two YAML example configuration is the RGB support (ceiling20 = with RGB LEDs, ceiling22 = without RGB LEDs). You should use the `ceiling22` as starting point because your device is CWWW only.

syssi, 2022-09-21:
Could you provide a photo of your ceiling lamp (LED driver board) without the WiFi daughter board attached? I would like to see the socket labels on the mainboard.

The photo should show something like this:https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/ylxd50yl/003.jpg

B48D81EFCC, 2022-09-21:
Hi,

yes I can and will provide.
But I can tell you already: It looks exactly like your photo and the labels are the same (R G B is on my board as well).

syssi, 2022-09-21:
Do you agree to add your photos to the project under Apache 2.0 license? The pull request of ceiling22 didn't provide any photos. So that's the chance to illustrate/improve the ceiling22 support.

B48D81EFCC, 2022-09-21:
Yes, sure. Feel free to add de photos to the project or do you want me to create a PR?

syssi, 2022-09-21:
I'm happy about a contribution too. :-)

syssi, 2022-09-21:
Feel free to close this issue as soon your setup is up and running.

B48D81EFCC, 2022-09-21:
Hi,

with the current documented pinout for ceilling22, I was not able to flash the firmware.
TXD = TP4
RXD = TP5

The correct pinout is
TXD = TP5
RXD = TP3

I will create another PR

B48D81EFCC, 2022-09-21:
I was able to flash the ceiling22 firmware / configuration.

Looks very well so far! I am very happy.

closed 2022-09-21 by B48D81EFCC