syssi/esphome-yeelight-ceiling-light #61: YLLD01YL dissasembly

adrez99, 2022-12-04:
I have the Yeelight Floor lamp and would like to convert it to esphome, but I can't figure how to dissasembly it. Does anybody has any clue how to do it? Thanks

syssi, 2023-02-05:
This device looks like a Staria Bedside Lamp, correct? Please check the underside of the stand. Does it looks like this?

https://raw.githubusercontent.com/syssi/esphome-yeelight-ceiling-light/main/images/ylct02yl/001.jpg

adrez99, 2023-02-05:
It looks like a larger Staria Bedside Lamp, but it's quite different:
[image]

I think the circuit board is inside the lamp, but haven't looked at it anymore.
Thank for your answer anyway

syssi, 2023-02-05:
If you know the miIO security token you could use `miiocli device info` to retrieve the device info messages. It should expose it's a ESP32 or Mediatek device.

closed 2023-06-25 by syssi