syssi/esphome-yeelight-ceiling-light #62: Custom firmware will avoid connecting drops?

helgek, 2022-12-08:
Hi, 

thanks so much for this contribution! 

I wanted to ask if your firmware would do a better job with making the device accessible again if it loses wifi connection in between? With the original firmware the lights once losing connection would not be accessible anymore via LAN Control (I use Home Assistant integration). I don’t know if they actually lose connection, maybe it’s another issue within the firmware itself. When this happens I need to power cycle the device to be able to access it again.

Thanks,
Helge

syssi, 2022-12-08:
If I reboot my WiFi access point my ESPHome based Yeelights are unavailable for a few seconds and recover automatically. No need to reboot here yet. It's hard to predict this also applies to your setup. ;-)

helgek, 2022-12-15:
@syssi thank you for the hint :) Unfortunately it doesn't work in my case. The lamp loses it's connection to the access point and also when I reboot my access point it won't reconnect. When turn it off and on it will reconnect. That's why I'm wondering if your custom firmware would handle wifi connection differently. Or maybe my device's software hangs. Forgot to try if I would still be able to control it with the BLE remote, but I think to recall that it also didn't respond to the remote anymore once it had disconnected from the wifi as well.

syssi, 2022-12-16:
If you make a backup of the stock firmware you could go back to the stock firmware if ESPHome doesn't work well for you. At the moment I expect the custom firmware will solve all your issues. Only the BLE remote control doesn't work out of the box.

syssi, 2022-12-16:
Which Yeelight model do you own?

Jamjon, 2022-12-20:
@syssi did you try the new BL proxy from ESPHome? Is it also boot-looping? This would male it possible to also connect the remote via HA.

syssi, 2022-12-20:
@Jamjon Could you create a new issue? Your question is off-topic.

Jamjon, 2022-12-20:
sure sorry

closed 2023-02-05 by syssi

helgek, 2023-02-05:
Short update jfyi: Though I haven't found time yet to further look into flashing custom firmware I have had good experiences with my ceiling lamps since I blocked the internet connection/cloud server access. I didn't have connections drops for several weeks, only one lamp froze yesterday (but this could be a different issue than a connection issue). My first candidate for flashing will certainly be my Meteroite/Pendant Light which always loses connection after a few hours despite blocked internet connection (known issue for many people as I found out studying the yeelight forum). My two Arwen 550/450 seem to be generally much more stable though.

syssi, 2023-02-05:
If you need support / if you have trouble to flash your first device please let me know! I'm pretty sure you will be surprised how awesome ESPHome performs in these devices if you don't depend on the BLE remote control.

helgek, 2023-02-05:
Thank you so much!! Since I'm setting everything up in my new place with Home Assistant and mostly Zigbee devices I will not rely on the BLE remote control. There's plenty of great remote control devices with Zigbee and so this will not hinder me from flashing :) Thanks again for your kind offer! It'll probably happen later this year since I have too many other projects that I need to address first but it will definitely happen at some point in time this year when I'm going to complete the setup of the new apartment.

helgek, 2023-03-20:
@syssi Today I felt motivation and capacity to start looking into flashing my first yeelight victim, the meteorite pendant light. I already found out how to disassemble it and so I tried then to find out what tools I would need for actually flashing the esp32 board. I ordered them with the help of [this page](https://tasmota.github.io/docs/Getting-Started/#needed-hardware). The tools will arrive later this week.

Also I just found out that there's also Tasmota firmware for the meteorite. Is there any substantial difference between your firmware files and Tasmota firmware?

syssi, 2023-03-21:
If you use Home Assistant the ESPHome firmware is probably the better choice because a ESPHome node can be integrated into HA without any configuration. I guess the stability and availability will be probably the same. You can switch/migrate between these both worlds whenever you want (via OTA). So just choose the easiest way for you.

helgek, 2023-03-21:
@syssi thank you very much for the explanation! Wish me luck that I don't break the board :D Will of course follow-up on the outcome and how far I've managed myself to flash the device. I found [this instruction of yours](https://github.com/syssi/esphome-yeelight-ceiling-light/issues/14), it looks like it would make sense to use it as a basis for my adventure.

helgek, 2023-09-07:
@syssi finally I wanted to follow-up on the outcome of my flashing efforts  :)

I wasn't very successful with my soldering efforts and killed the meteorite microcontroller. Fortunately I found a used meteorite lamp online for a decent price and so I had one more try by using the microcontroller from this used pendant light. But eventually I found a way to flash the microcontroller without soldering! This took me hell of a long research online but with the solution I discovered - you won't believe me probably - I even managed to flash my two Arwen 550S and one Arwen 450C without soldering! 

Did I make you curious how I managed to do this? I hope! Well, here's the solution (and no, I didn't ask somebody else to do it for me :D):

The solution is called PCBite. Some genius engineers from Sweden (I assume) have invented this game changing tool, one of the most outstanding innovations I have seen in the area of tools: https://sensepeek.com/ 

With this tool even a soldering moron like me can accomplish the flashing exercise which seemed to be a goal out of reach especially when I faced the PCBs of the Arwen lamps (the Arwen 450C had an additional challenge ready for me since you have to cut the plastic cover to actually be able to reach inside; and then the locking mechanism revealed that this lamp was designed this way that there's no way to open it again except by cutting into the cover of the lamp; in addition the PCB was very difficult to hold for the flashing exercise since it wasn't as flexible to be moved like it was the case with my two Arwen 550S ).

I have no clue how people soldered e.g. the Arwen 450C since it's seems impossible to reach some of the pins for soldering due to how they're squeezed between two PCBs. Even with PCBite it was still an artistic exercise to place some of the probes to hold on the pins of the microcontroller. But luckily it worked out in the end!

Now with 4 ESPhome flashed yeelight lamps I can report that they now all work 100 percent reliable, no connection drops, no disappearing, no issues! And it has been like this for several months now (I had flashed them about 3 months ago)!

For everyone who's interested. I ordered these two PCBite sets:

Sensepeek 4005 SP10 
Sensepeek 6003 PCBite 

In addition I had bought helping arms, like these ones: https://www.amazon.de/gp/product/B09V7SZ6XB/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Of course this is quite an investment if you own only one lamp, but even for me it took some consideration to actually go this path. I didn't want to return the PCBite after I had flashed my lamps since this would just have been not fair towards this company. Considering the great quality of the product and its uniqueness I think it's still very fairly priced. For me it was clear that their was no alternative to the yeelight lamps in my apartment, so I eventually decided to do it.

Thank you again Sebastian and everyone else who contributed to this project! I think if you put the information about PCBite  in the Readme you will make many people happy and motivate many more people to go the flashing path for their yeelight lamps :)

Here are two photos showing my successful flashing efforts with PCBite (Meteorite/Crystal Pendant light and Arwen 450C):

[image]
[image]

helgek, 2023-09-27:
@syssi I don't know if you had seen my previous comment. Just wanted to suggest to feature the PCBite solution in the Readme to give more people the chance to flash their Yeelight lamps.

syssi, 2023-09-28:
@helgek Could you publish your comment + photos above here: https://github.com/syssi/esphome-yeelight-ceiling-light/discussions/

Please pick the `Show and tell` section. I will add a hint to the README as soon as possible.

helgek, 2023-09-28:
@syssi Of course. I just posted it to this section. This is the direct link: https://github.com/syssi/esphome-yeelight-ceiling-light/discussions/72