syssi/esphome-yeelight-ceiling-light #63: YLXD41YL bootloop in esphome 2022.12

bedson87, 2022-12-17:
Hi, I have a problem with the lamp after updating to esphome version 2022.12 it goes into bootloop. Anyone else noticed problems? The config and serial logs are attached. Unfortunately, I had to disassemble the lamp and flash the 2022.11.5 version.
[serial_logs_reboot.txt](https://github.com/syssi/esphome-yeelight-ceiling-light/files/10251879/serial_logs_reboot.txt)

```
# merged configuration of:
# - https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceiling15.yaml
# - https://github.com/jaddel/ESPHome-Configurations/blob/master/Devices/Ceiling%20Yeelight%20YLXD41YL/ceiling_yeelight_ylxd41xl.yaml

substitutions:
  name: kuchnia-yeelight-ceiling
#  external_components_source: github://syssi/esphome-yeelight-ceiling-light@main

esphome:
  name: ${name}

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.10.21
  fast_connect: on
  manual_ip:
    static_ip: 192.168.10.21
    gateway: 192.168.10.1
    subnet: 255.255.255.0

ota:
  password: "XXXXXXX"
api:
logger:

sensor:
  - platform: adc
    pin: GPIO36
    name: "${name} adc1"
    attenuation: 11db
  - platform: adc
    pin: GPIO32
    name: "${name} adc2"
    attenuation: 11db
#dodatkowe sensory
  - platform: uptime
    name: "${name} Last Restart"
    update_interval: 300s
    device_class: "timestamp"
    unit_of_measurement: ""
    filters:
      - lambda: |-
          auto time = id(homeassistant_time).now();
          if (time.is_valid()) {
            return int(time.timestamp) - int(x);
          } else {
            return {};
          }
  - platform: wifi_signal
    name: "${name} WiFi Signal"
    update_interval: 300s
    device_class: "signal_strength"

output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    frequency: 4882Hz
  - platform: ledc
    pin: GPIO21
    id: output_cold
    frequency: 4882Hz
  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    frequency: 19531Hz

light:
  - platform: monochromatic
    name: "${name}-nightlight"
    id: night_light
    output: output_nightlight
    gamma_correct: 0.7
    on_turn_on:
      - light.turn_off: ceiling_light
  - platform: cwww
    name: "${name}"
    id: ceiling_light
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true
    gamma_correct: 0.4
    on_turn_on:
      - light.turn_off: night_light

time:
  - platform: homeassistant
    id: homeassistant_time
```

syssi, 2022-12-17:
I will try to reproduce the issue by flashing your YAML (with single core mode enabled) to an generic ESP32 board. I haven't updated my Yeelights to 2022.12 yet.

bedson87, 2022-12-17:
Thank you!  I tried to flash v2022.12 by uart but unfortunately also bootloop.

syssi, 2022-12-17:
Did you use 2022.12 or 2022.12.1? I cannot reproduce the issue on a generic ESP32.

syssi, 2022-12-17:
If you are able to reproduce the issue could you use `esphome logs your-yeelight.yaml` to monitor the boot loop? The ESPHome command line tool is able to decode the stack trace above on the fly. I would like to know the decoded defails of

```
Backtrace:0x40082036:0x3ffc1e800x40088399:0x3ffc1ea0 0x4008e3e6:0x3ffc1ec0 0x400ecfdb:0x3ffc1f30 0x40141324:0x3ffc1f70 0x401413e9:0x3ffc1fa0 0x40120d56:0x3ffc1fc0 0x401214c9:0x3ffc1fe0 0x4011fdac:0x3ffc2000 0x4008fd61:0x3ffc2020
```

bedson87, 2022-12-17:
I tried 2022.12.  ESP the one that was in the lamp (maybe the Chinese put something weird in there ). I will test v2022.12.1 when I have some time to disassemble the lamp ;-)
I haven't tried to log via eshome cli -  I'll try.

bedson87, 2022-12-18:
I tried to install via OTA 2022.12.1 - bootloop (uart logs). I launched a separate eshome instance in docker and connected via UART with lamp - eshome cli logs also attached.

```
[13:11:26][D][esp-idf:000]: 
[13:11:26]
[13:11:26][D][esp-idf:000]: I (847) wifi_init: rx ba win: 6
[13:11:26]
[13:11:26][D][esp-idf:000]: I (856) wifi_init: tcpip mbox: 32
[13:11:26]
[13:11:26][D][esp-idf:000]: I (857) wifi_init: udp mbox: 6
[13:11:26]
[13:11:26][D][esp-idf:000]: I (867) wifi_init: tcp mbox: 6
[13:11:26]
[13:11:26][D][esp-idf:000]: I (867) wifi_init: tcp tx win: 5744
[13:11:26]
[13:11:26][D][esp-idf:000]: I (877) wifi_init: tcp rx win: 5744
[13:11:26]
[13:11:26][D][esp-idf:000]: I (887) wifi_init: tcp mss: 1440
[13:11:26]
[13:11:26][D][esp-idf:000]: I (897) wifi_init: WiFi IRAM OP enabled
[13:11:26]
[13:11:26][D][esp-idf:000]: I (898) wifi_init: WiFi RX IRAM OP enabled
[13:11:26]
[13:11:26][D][esp-idf:000]: I (909) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
[13:11:26]
[13:11:26][D][esp-idf:000]: W (918) phy_init: failed to load RF calibration dat
[13:11:26]abort() was called at PC 0x400ecfdb on core 0
[13:11:26]
[13:11:26]
[13:11:26]Backtrace:0x40082036:0x3ffc1e900x40088399:0x3ffc1eb0 0x4008e3e6:0x3ffc1ed0 0x400ecfdb:0x3ffc1f40 0x40141324:0x3ffc1f80 0x401413e9:0x3ffc1fb0 0x40120d56:0x3ffc1fd0 0x401214c9:0x3ffc1ff0 0x4011fdac:0x3ffc2010 0x4008fd61:0x3ffc2030 
WARNING Found stack trace! Trying to decode it
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
[13:11:26]
[13:11:26]
[13:11:26]
[13:11:26]
[13:11:26]ELF file SHA256: 67087a46284a5974
[13:11:26]
[13:11:26]Rebooting...
[13:11:26]ets Jun  8 2016 00:22:57
[13:11:26]
[13:11:26]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:11:26]configsip: 0, SPIWP:0xee
[13:11:26]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:11:26]mode:DIO, clock div:2
[13:11:26]load:0x3fff0018,len:4
[13:11:26]load:0x3fff001c,len:1044
[13:11:26]load:0x40078000,len:8896
[13:11:26]load:0x40080400,len:5828
[13:11:26]entry 0x400806ac
[13:11:27]I (308) cpu_start: Pro cpu up.
[13:11:27]I (308) cpu_start: Single core mode
[13:11:27]I (316) cpu_start: Pro cpu start user code
[13:11:27]I (316) cpu_start: cpu freq: 160000000
[13:11:27]I (316) cpu_start: Application information:
[13:11:27]I (321) cpu_start: Project name:     kuchnia-yeelight-ceiling
[13:11:27]I (327) cpu_start: App version:      2022.12.1
[13:11:27]I (332) cpu_start: Compile time:     Dec 18 2022 13:39:57
[13:11:27]I (338) cpu_start: ELF file SHA256:  67087a46284a5974...
[13:11:27]I (344) cpu_start: ESP-IDF:          4.4.2
[13:11:27]I (349) heap_init: Initializing. RAM available for dynamic allocation:
[... 910 lines cut ...]
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
/root/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-addr2line: '/config/.esphome/build/kuchnia-yeelight-ceiling/.pioenvs/kuchnia-yeelight-ceiling/firmware.elf': No such file
[... cut ...]
[13:11:32]I (344) cpu_start: ESP-IDF:          4.4.2
[13:11:32]I (349) heap_init: Initializing. RAM available for dynamic allocation:
[13:11:32]I (356) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

```
[logs_cli_esphome.txt](https://github.com/syssi/esphome-yeelight-ceiling-light/files/10253881/logs_cli_esphome.txt)
[serial_logs_reboot.txt](https://github.com/syssi/esphome-yeelight-ceiling-light/files/10253883/serial_logs_reboot.txt)

My other ESP32s work after updating to 2022.12 without a problem (esp32-wroom).

tbrasser, 2023-01-13:
Still having this issue when ota'ing from HA addon, even after cleaning all related files... 
Well, here we go again opening and flashing

syssi, 2023-01-14:
You can find some details about the root cause here: https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/104

I've updated the example configurations of the project accordingly to avoid this issue. With the next ESPHome release the issue should be fixed without touching the config YAML.

My recommendation: Don't update your Yeelights to ESPHome 2022.12 if possible and wait for 2023.1.

fluffymadness, 2023-02-09:
I can confirm that it is not a good idea to update now. my mi desklamp pro has the same behavior. I flashed the file with the fix via esphome web flasher and I am in a constant bootloop no matter what.

tbrasser, 2023-02-16:
Fixed in 2023.2.x! 
You can remove the workaround. 
Flashed it with dev without workaround from bootloop state, then ota'd it to 2023.2.1 via my main esphome (dashboard), still working!

Don't forget to clean build files up front!

syssi, 2023-02-16:
Thanks! I will give it a try this weekend on an easily accessible/recoverable lamp. :-)

bedson87, 2023-02-16:
My lamp updated without a fix to 2023.2 without any problems.

closed 2023-06-25 by syssi