syssi/esphome-yeelight-ceiling-light #64: Anyone expirience BL proxy from ESPHome on yeelight?

Jamjon, 2022-12-20:
Like the title already say, i was wondering if anybody already tried the new BL proxy feature from ESP home on one of this lights?
Since the Xiaomi BLE creates boot-loops i was wondering if the new Proxy is working better.

yvolchkov, 2022-12-26:
I just did. Lost connectivity to my lamp completely. Doesn't even go to the safe mode, need to disassemble and refresh :(

yvolchkov, 2022-12-26:
turns out if I disable encryption in my config, lamp even boots, and responds to pings, but refuses connection to 6053 port.

Jamjon, 2022-12-26:
thank you for testing! hope its not too much hustle to bring it down and up again.

ExPeacer, 2022-12-27:
Will not work, since in Yeelights they use a stripped version of ESP32 - which uses only 1 core of the 2, so simply don't have enough resources to handle the ESPHome code, and the BT too.

syssi, 2022-12-27:
@ExPeacer You are right about the number of CPU cores but are you sure about the limitations? I'm aware of some people running a single core ESP32 with ESPHome and bluetooth in general. It's not as smooth/solid as having two cores but it's possible.

yvolchkov, 2022-12-28:
it's also enough resources for the original yeelight firmware. Perhaps it is possible to tune FreeRtos to run both stacks.

Also my board response to pings, when it boots WITH bluetooth stack enabled. But it doesn't respond to api. Unfortunately I don't have tools to boot it properly with serial attached, to see what is going on.

syssi, 2022-12-28:
This is the workaround to avoid the boot loop (`Task watchdog got triggered. The following tasks did not reset the watchdog in time`): https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_fancl5.yaml#L8-L13

An unresponsible native API and the `assert failed: s_prepare_reserved_regions memory_layout_utils.c:100` looks like other issues.

thatslolo, 2023-11-22:
As far as I'm aware Yeelight uses ESP32s with only 2MB of flash. This issue can be circumvented when using a correct partition table. ESPHome won't flash since the fw would be too big for the two app partitions with bt proxy enabled.

```
esphome:
  platformio_options:
    board_build.partitions: "../../../boards/2mb_partition.csv"
```

partition table csv:

```
# Name,   Type, SubType, Offset,   Size, Flags
nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
app0,     app,  ota_0,   ,        0xEA640,
app1,     app,  ota_1,   ,        0xEA640,
```

When trying to flash with bt proxy and partition table:

```
Error: The program size (1327701 bytes) is greater than maximum allowed (960064 bytes)
RAM:   [==        ]  15.4% (used 50420 bytes from 327680 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  138.3% (used 1327701 bytes from 960064 bytes)
```

syssi, 2025-01-24:
https://github.com/syssi/esphome-yeelight-ceiling-light/pull/96#issuecomment-2600974154

@mdvorak was able to avoid the boot loop by shifting the setup priority of the BLE tracker:

```
esp32_ble_tracker:
  setup_priority: -900  # This is important, otherwise single-core ESP won't boot
  scan_parameters:
    active: false
```

Let's give it a try and spin up the BLE proxy. :-)