syssi/esphome-yeelight-ceiling-light #65: Yeelight Meteorite (ceiling10): replacing board

PsychoRS, 2023-01-23:
Hi,

Due to poor soldering skills, a friend of mine has burned the board of his Yeelight Meteorite.

Is it possible to replace it with a cheap ESP32-WROOM board? Luckily enough, the Meteorite has the connections for the board labeled into the case. so...

It just wire GND, +5V, PC, PW, NL, R, G, B and ON/OFF to the corresponding GPIO on the cheapo board based on the yaml configuration?

What are the two pins (labeled GND and +50V) for in the OEM board, left side? And the +50V one on the RGB plug at right side?

[image]

Thanks

syssi, 2023-01-23:
You could replace the ESP32 of the driver board but it's hard to replace the full driver board because there is an LED driver next to the ESP.

syssi, 2023-01-23:
Let's call the `GND` and `+50V` pin on the left `input`. The jack on the right is the `output`. I assume each Mosfet is used to drive a color (`R`, `G`, `B`).

PsychoRS, 2023-01-23:
> Let's call the `GND` and `+50V` pin on the left `input`. The jack on the right is the `output`. I assume each Mosfet is used to drive a color (`R`, `G`, `B`).

Thanks, so It looks that its not possible then...

syssi, 2023-01-23:
What burned exactly of the driver board?

PsychoRS, 2023-01-23:
> What burned exactly of the driver board?

Its an assumption, he just stripped the 3.3v track in the back of the board trying to solder the wires to the points for flashing, and now the lamp doesn't work

syssi, 2023-01-28:
Do you agree to close the issue?

closed 2023-02-05 by syssi