syssi/esphome-yeelight-ceiling-light #70: Yeelight Light Strip1S on/off button does not work

doe1097, 2023-08-23:
Hi,
I successfully flashed a 1S light strip which was literally buried in a closet for the past two years, thank you! The light strip now works with Homeassistant, but with the new firmware the hardware on/off button stopped working. Is this a known issue or did I accidentally break the button, when I (forcefully) opened the case?

Best,
H.

syssi, 2023-08-23:
The button isn't implemented yet. Could you try to identify the GPIO attached to the button?

doe1097, 2023-08-23:
Sure. Can I do this by using the Log (and how)? I set the logger component to `level: DEBUG`, but when I press the button nothing shows up in the logs.

syssi, 2023-08-23:
You have two options:

1. Unplug the LED strip and use a multimeter:Measure the resistance between GND and all GPIOs. Press the button and measure the resistance again. Pressing the button should change the resistance of the dedicated GPIO. If you don't see an effect (if you press the button) repeat the experiment. This time measure the resistance between VCC and all GPIOs.
2. Extend your YAML by a binary sensor per GPIO (see https://esphome.io/components/binary_sensor/gpio.html). Avoid the already mentioned GPIOs at the YAML. Start the ESP, press the button multiple times. You should see some feedback in the logs now.

doe1097, 2023-08-23:
Looks like `GPIO19` is what we are looking for

syssi, 2023-08-23:
Good job! I will provide some glue code soon to toggle the light entity per button press.

syssi, 2024-01-20:
I've extended the YAML configuration. See https://github.com/syssi/esphome-yeelight-ceiling-light/commit/6546f196f42a2ecef8d2b7757bb680acf8b25af9

Please give it a try!

commit 0fb796c7 (2024-01-20): Toggle LED strip on button press (Closes: #70) (#75)
closed 2024-01-20 by syssi