syssi/esphome-yeelight-ceiling-light #73: Issues with restoring to the previous state after updates and power outages.

WiZaR5789, 2024-01-15:
Hi, do you have any thoughts on how to make an ESP32 (flashed with ESPHome) retain its state after a restart? Currently, with each update or power outage, I have to set the brightness and color from the beginning. If this is a straightforward question and I should know the answer, please point me in the direction where I can find it. Below, I'm providing YAML with code for ESPHome, which I am using.

yldl01yl:
'''
substitutions:
  name: yldl01yl

esphome:
  name: ${name}

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "yldl01yl Fallback Hotspot"
    password: "************"

ota:

logger:
  level: DEBUG

api:

output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    power_supply: power
    min_power: 0.13
    max_power: 0.77
    frequency: 1220Hz
    zero_means_zero: true
  - platform: ledc
    pin: GPIO21
    id: output_cold
    power_supply: power
    frequency: 1220Hz
    zero_means_zero: true
    min_power: 0.13
    max_power: 0.76

  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    power_supply: power

  - platform: ledc
    pin: GPIO33
    id: output_red
    power_supply: power
  - platform: ledc
    pin: GPIO26
    id: output_green
    power_supply: power
  - platform: ledc
    pin: GPIO27
    id: output_blue
    power_supply: power

power_supply:
  - id: power
    pin: GPIO22
    enable_time: 0s
    keep_on_time: 0s

light:
  - platform: monochromatic
    name: "${name} night light"
    id: night_light
    output: output_nightlight
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: ceiling_light
  - platform: cwww
    name: "${name} ceiling light"
    id: ceiling_light
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: night_light
  - platform: rgb
    name: "${name} ambient light"
    red: output_red
    green: output_green
    blue: output_blue
    gamma_correct: 0
'''

ylxd013-b:
'''
substitutions:
  name: ylxd013-b

esphome:
  name: ${name}

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ylxd013-b Fallback Hotspot"
    password: "*************"

ota:

logger:
  level: DEBUG

api:

output:
  - platform: ledc
    pin: GPIO19
    id: output_warm
    power_supply: power
#   min_power: 0.13
    max_power: 0.50
  - platform: ledc
    pin: GPIO21
    id: output_cold
    power_supply: power
#   min_power: 0.13
    max_power: 0.96

  - platform: ledc
    pin: GPIO23
    id: output_nightlight
    power_supply: power

  - platform: ledc
    pin: GPIO33
    id: output_red
    power_supply: power
  - platform: ledc
    pin: GPIO26
    id: output_green
    power_supply: power
  - platform: ledc
    pin: GPIO27
    id: output_blue
    power_supply: power

power_supply:
  - id: power
    pin: GPIO22
    enable_time: 0s
    keep_on_time: 0s

light:
  - platform: monochromatic
    name: "${name} night light"
    id: night_light
    output: output_nightlight
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: ceiling_light
  - platform: cwww
    name: "${name} ceiling light"
    id: ceiling_light
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true
    gamma_correct: 0
    on_turn_on:
      - light.turn_off: night_light
  - platform: rgb
    name: "${name} ambient light"
    red: output_red
    green: output_green
    blue: output_blue
    gamma_correct: 0
'''

syssi, 2024-01-16:
You are looking for a feature which is missing in ESPHome as far as I know. At the moment only the state (on/off) of a light entity can be restored easily. This doesn't include the brightness and color/colortemp.

syssi, 2024-01-16:
I've extended all my automations by the expected default (colortemp + brightness):

```
  action:
    - service: light.turn_on
      entity_id: light.nursery
      data:
        brightness_pct: 100
        color_temp: 370
```

In this case the previous state of the light doesnt matter anymore. Sometimes I've a `input_number` per room to be able to pick up some easily configurable defaults:

```
  action:
    - service: light.turn_on
      entity_id: light.nursery_nightlight
      data_template:
        brightness_pct: "{{ states('input_number.nursery_light_brightness_double_click')|int(1) }}"
        color_temp: 370

# ...

    - service: light.turn_on
      entity_id: light.nursery
      data_template:
        brightness_pct: "{{ states('input_number.nursery_light_brightness_single_click')|int(100) }}"
        color_temp: 370
```

syssi, 2024-01-16:
This is another workaround to manipulate some settings on boot to avoid unloved states:

```
# https://github.com/syssi/philipslight
esphome:
  name: "${hostname}"
  platform: ESP8266
  board: esp_wroom_02
  board_flash_mode: dout
  includes:
    - common/custom_components/xiaomi_light.h
  on_boot:
    then:
      - light.turn_on:
          id: bulb
          brightness: "$brightness_on_boot"
          color_temperature: 333mireds
```

Please don't copy&paste the complete code block. The `on_boot` section is the relevant part for you!

WiZaR5789, 2024-01-16:
It's some idea, thanks. I'll probably go with the second option.

closed 2024-01-16 by WiZaR5789