syssi/esphome-yeelight-ceiling-light #77: Improve boot times

maumaumaumaumau, 2024-02-14:
Hi,
I noticed, that yeelight lamps with standard firmware are starting blazing fast, where as esphome flashed lamps require 1-2 seconds to turn on.
Not only that is a a hassle for presence sensors, but also guests think lamps are broken, if lamps don't turn on instantly.

Is there a way to improve that behavior?

BR
mau

maumaumaumaumau, 2024-02-14:
Yes it is, but it's a project for the future. I was just wondering, if we can overcome that issue, because that's the only real downside compared to OFW.

syssi, 2024-02-14:
Do you expect the last state of the device is retrieved from the Home Automation first? If the last state doesn't matter you could add a `turn_on` statement to the `on_boot` section to speed things up. Also turning off transitions will feel like better responsibility.

maumaumaumaumau, 2024-02-14:
I experimented a little.
By commenting out all restore_mode: lines and adding the following, did not improve the situation at all:

  on_boot:
    priority: 750
    then:
      - light.turn_on: 
          id: ceiling_light
          brightness: 100%   
          cold_white: 0%
          warm_white: 100%

tmp-hallenser, 2024-04-01:
I'm facing the same issues. Any solution in the meantime?

maumaumaumaumau, 2024-04-11:
No sorry, I tried the priority and the transition option.
Maybe I did something wrong, as I'm not too experienced with esp programming.

B48D81EFCC, 2024-07-06:
How is the experience when:
1. you turn on the light
2. Wait for it until it is turned on
3. Turn off and directly turn on again (max. pause around 1 second)

In my case the light turns on directly without any delay.
If the pause between "off" and "on" is too long, I have a delay turning on the lamp again.
I am wondering what cause this behavior.

B48D81EFCC, 2024-07-13:
I replaced the built-in board with an ESP32 S3 WROOM (T-ETH-Lite ESP32-S3).
Using the S3 board, the lamp is almost instantly on after power cycle. MAYBE the stock firmware is slightly faster. I can't compare it side by side. But with the S3 it is MUCH faster than with the built-in board.

B48D81EFCC, 2024-07-21:
Finally I decided to use a Seeed Studio XIAO ESP32S3 and replaced the build-in board.
The lamp turns on without delay now.

kosmos342, 2024-12-21:
I reduced delay on power on with:
```
esp32:
  ...
  framework:
    ...
    sdkconfig_options:
      ...
      CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON: y
      CONFIG_BOOTLOADER_LOG_LEVEL_NONE: y
      CONFIG_LOG_DEFAULT_LEVEL_NONE: y
...
logger:
  level: NONE
  baud_rate: 0
...
light:
  - platform: ...
    ...
    default_transition_length: 0s
```
and restore_mode or boot priority 750 to turn on (I use `restore_mode: RESTORE_AND_ON`)
I'm new to esphome, so some of the log parameters might be redundant (I haven't researched this)

B48D81EFCC, 2024-12-24:
I just stumbled over a esphome feature request to disable serial logs by default due to performance improvments: https://github.com/esphome/feature-requests/issues/2993

maumaumaumaumau, 2025-01-04:
> I just stumbled over a esphome feature request to disable serial logs by default due to performance improvments: [esphome/feature-requests#2993](https://github.com/esphome/feature-requests/issues/2993)

I can confirm, it really speeds up the boot time.

syssi, 2025-01-04:
Feel free to contribute the improvement. I would like to see the change as part of the examples.

maumaumaumaumau, 2025-01-05:
I literally just took the code:
`logger:
  level: DEBUG
  baud_rate: 0`

In my opinion the lamp does not turn on immediately (as in stock FW), but it definitely turns on twice as fast as without the change.

closed 2025-01-18 by syssi