syssi/esphome-yeelight-ceiling-light #78: Add support for yeelink.light.lamp10

mrumpf, 2024-03-02:
miiocli device --ip ... --token ....
Model: yeelink.light.lamp10
Hardware version: esp32
Firmware version: 2.0.6_0013

helgek, 2024-03-02:
There is support already for this lamp. You can also see this thread https://github.com/syssi/esphome-yeelight-ceiling-light/discussions/72 where I added a photo of how I flashed it.

mrumpf, 2024-03-03:
Thanks for the hint!

However, as I had no sophisticated equipment, I had to improvise ...

I was able to solder 3 wires to the board to some larger pads:
- RX - green
- TX - blue
- GPIO0 - black

And user alligator clips with simple pins to connect GND and 3.3V from my USB to serial adapter to the ESP32. 

[image]

So far I was able to read the 4MB original firmware from the lamp. This takes several minutes and you need a steady hand :) When you move the pins even slightly, the transfer of the original firmware will fail.

helgek, 2024-03-03:
Congrats! :D When you managed the reading part you will certainly manage the writing part as well. Another way you could probably achieve this without soldering and without investing into the PCBite equipment would be to e.g. use the pins you used or to e.g. use acupuncture needles. One could tie some thick foam onto the esp32 chip and push the needles through the foam. The foam would hold them in place. Just another idea that just came to my mind. Nevertheless I'm glad I invested into the PCBite since I've used it to flash six lamps so far. The other lamp's esp32s were more difficult to access, so the PCBite was my rescue.

mrumpf, 2024-03-03:
Related: #61 
[image]

[image]

helgek, 2024-03-03:
Ah sorry, I just realized you're not trying to flash a meteorite pendant light but this lamp also identifies as yeelink.light.lamp10. The pendant light doesn't have manual controls on the lamp itself. The bedside lamp has such controls included: https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_lamp9.yaml

mrumpf, 2024-03-03:
I also have the bedside lamp, but it identifies as yeelink.light.bslamp2. I already flashed that one with the guide from here:
https://devices.esphome.io/devices/Xiaomi-Mijia-Bedside-Lamp-2
This has the same control panel as the Staria floor lamp yeelink.light.lamp9.

mrumpf, 2024-03-09:
OK, I managed to solder the necessary wires directly to the ESP32 since the reverse engineering is taking much more time than expected. I first flashed the firmware from [esphome-xiaomi_bslamp2](https://github.com/mmakaay/esphome-xiaomi_bslamp2), but I could not turn on the light. I suspected that the I2C bus was connected to different GPIO pins or that the I2C address of the front panel was different. So I flashed some I2C scanner tool to the ESP32 and used the same GPIO of the [esphome-xiaomi_bslamp2](https://github.com/mmakaay/esphome-xiaomi_bslamp2):
- GPIO21 SDA
- GPIO19 SCL

That revealed that there is an I2C device discovered with the address 0x50:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Then I modified the [example.yaml](https://github.com/mmakaay/esphome-xiaomi_bslamp2/blob/dev/example.yaml) by adding overwriting the I2C address of the front-panel from the core.yaml file:
```
xiaomi_bslamp2:
  front_panel:
    address: 0x50
```
After flashing it to the ESP32 of the lamp, I received messages from the front-panel like this:
```
[E][xiaomi_bslamp2:149]: Front panel I2C event error:
[E][xiaomi_bslamp2:150]:   Error: prefix is not 04:04:01:00
[E][xiaomi_bslamp2:151]:   Event: [0a:01:03:00:00:00:02]
[E][xiaomi_bslamp2:152]:   Parsed part: n/a
[E][xiaomi_bslamp2:153]:   Parsed event type: n/a
[W][xiaomi_bslamp2:255]: Skipping unsupported message from front panel
```
That is a good sign because it means that the I2C address 0x50 is the address of the front-panel. But the events from the fornt-panel seem to be different and that the firmware cannot interpret the content (yet).
Receiving messages with the button press also means that the same GPIO16 is used for the trigger to read data from the front-panel.

closed 2026-09-12 by mrumpf