syssi/esphome-yeelight-ceiling-light #86: Yeelight Ceiling Light 300C yaml

rbalaev, 2024-08-02:
[image]

Hello, I can't find which config file to use, there is no such file **yeelight.light.ceil20**

syssi, 2024-08-02:
Could catch. There are some details missing. Could you provide a photo of your Yeelight mainboard?

I guess this configuration should work: https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceil26.yaml

rbalaev, 2024-08-02:
> Could catch. There are some details missing. Could you provide a photo of your Yeelight mainboard?
> 
> I guess this configuration should work: https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceil26.yaml

This is exactly the same lamp as in your description.

syssi, 2024-08-02:
https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceiling20.yaml is the correct YAML configuration for your lamp.

rbalaev, 2024-08-09:
Can you correct the description?

syssi, 2024-08-09:
Do you still use the original firmware? I would like to know the mDNS name of the stock firmware. Is the device advertised as `...ceil20` or `...ceiling20`?

rbalaev, 2024-08-09:
[image]

closed 2025-01-18 by syssi