syssi/esphome-yeelight-ceiling-light #88: ceiling1 aka YIXD05YI support

fgsfds, 2024-08-29:
YIXD05YI does seem to have ESP, but I barely have any experience with it, so I require someone's assistance. Thanks in advance.

[image]

[image]

syssi, 2024-08-29:
Could you provide some more photos of the mainboard? Do you have a continuity tester or multimeter to check the continuity between the pin header (4 pins) and the pads of the ESP32?

fgsfds, 2024-08-29:
I have a multimeter

[image]

[image]

syssi, 2024-08-29:
Please measure the continuity / resistance between the pins of the pin header and the ESP32. May you could label the corresponding pads here:

[image]

https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf

fgsfds, 2024-08-29:
What am I looking for? I one of the header pins, and there is resistant between multiple esp pins (like 10, 14 and 25).

fgsfds, 2024-08-29:
What am I looking for? I checked one of the header pins, and there is resistance between it and multiple esp pins (like 10, 14 and 25).

fgsfds, 2024-08-29:
I don't think any of these pins are TX/RX. That's what I got 

[image]

Two bottom pins have zero resistance between ground and 3v3

closed 2026-11-01 by syssi