syssi/esphome-yeelight-ceiling-light #89: YLXD41YL - mac address errors after 2024.10

bedson87, 2024-11-05:
Everything works fine o 2024.09. After updating to ver 2024.10 there are errors: 
`[D][esp-idf:000]: E (4412) system_api: Base MAC address from BLK3 of EFUSE version error, version = 77`

Yaml:
> substitutions:
> 
>   name: kuchnia-yeelight-ceiling
> 
>  
> 
> esphome:
> 
>   name: ${name}
> 
>   min_version: 2024.6.0
> 
>  
> 
> esp32:
> 
>   board: esp32doit-devkit-v1
> 
>   framework:
> 
>     type: esp-idf
> 
>     sdkconfig_options:
> 
>       CONFIG_FREERTOS_UNICORE: y
> 
>     advanced:
> 
>       ignore_efuse_mac_crc: true
>
>  
> 
> wifi:
> 
>   ssid: !secret wifi_ssid
> 
>   password: !secret wifi_password
> 
>   use_address: 192.168.10.21
> 
>   fast_connect: on
> 
>   manual_ip:
> 
>     static_ip: 192.168.10.21
> 
>     gateway: 192.168.10.1
> 
>     subnet: 255.255.255.0
> 
>  
> 
> ota:
> 
>   - platform: esphome
> 
>     password: "XXXXX"
> 
>  
> 
> api:
> 
>   encryption:
> 
>     key: "XXXXX"
> 
>  
> 
> logger:
> 
>  
> 
> sensor:
> 
>   - platform: adc
> 
>     pin: GPIO36
> 
>     name: "${name} adc1"
> 
>     attenuation: 12db
> 
>   - platform: adc
> 
>     pin: GPIO32
> 
>     name: "${name} adc2"
> 
>     attenuation: 12db
> 
> #dodatkowe sensory
> 
>   - platform: wifi_signal
> 
>     name: "${name} WiFi Signal"
> 
>     update_interval: 300s
> 
>     device_class: "signal_strength"
> 
>  
> 
> output:
> 
>   - platform: ledc
> 
>     pin: GPIO19
> 
>     id: output_warm
> 
>     frequency: 4882Hz
> 
>   - platform: ledc
> 
>     pin: GPIO21
> 
>     id: output_cold
> 
>     frequency: 4882Hz
> 
>   - platform: ledc
> 
>     pin: GPIO23
> 
>     id: output_nightlight
> 
>     frequency: 19531Hz
> 
>  
> 
> light:
> 
>   - platform: monochromatic
> 
>     name: "${name}-nightlight"
> 
>     id: night_light
> 
>     output: output_nightlight
> 
>     gamma_correct: 0.7
> 
>     on_turn_on:
> 
>       - light.turn_off: ceiling_light
> 
>   - platform: cwww
> 
>     name: "${name}"
> 
>     id: ceiling_light
> 
>     cold_white: output_cold
> 
>     warm_white: output_warm
> 
>     cold_white_color_temperature: 6500 K
> 
>     warm_white_color_temperature: 2700 K
> 
>     constant_brightness: true
> 
>     gamma_correct: 0.4
> 
>     on_turn_on:
> 
>       - light.turn_off: night_light

bedson87, 2024-11-05:
You can upload old fw via ota. But you can't add it to ha

kntsoft, 2024-11-16:
This is the same issue that https://github.com/esphome/issues/issues/6333 probably in the next release the ESPHome team will fix it.

You can fix it adding `ignore_efuse_custom_mac`

```yaml
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
      ignore_efuse_custom_mac: true
```

syssi, 2024-11-17:
I've applied the required setting to the examples to avoid some hassle. Thanks for pointing out this issue and providing a solution!

PR #91 (2024-11-17): Use ignore_efuse_custom_mac to be ESPHome 2024.10 compatible
Closes: #89
commit 63ce15f2 (2024-11-17): Use ignore_efuse_custom_mac to be ESPHome 2024.10 compatible (#91)

Closes: #89
closed 2024-11-17 by syssi