syssi/esphome-yeelight-ceiling-light #101: yeelink.light.ceil26 -- web_server got enabled after flashing to ESPHome 2025.3.0 and

vit5041, 2025-03-20:
Hi,
I've got strange issue - suddenly noticed that my yeelight lamp have got a web_server enabled while it's being commented in sketch.
I tried to flash firmware for several times and always get same result -- web_ server was enabled.

Then I noticed the following error after firmware loading:

[image]

After rollback ESPHome to 2025.2.2 the web_server has dissapeared. 
And the error has gone as well, didn't see it after initial firmware loading. But when tried to load firmware again -- saw the same error again. I am not sure if the error has anything to do with web_server component.. 
Do you have any idea why web_server become available when it's switched off in sketch?

syssi, 2025-03-21:
Could you clean up your build folder and give it another try?

vit5041, 2025-03-21:
> Could you clean up your build folder and give it another try?

I did it, flashed several times after clean, result was always same.

vit5041, 2025-03-21:
> If you hit the issue another time please archive the build folder and provide a copy.

I just thought that someone may meet it as well. I understand that it's strange. Later this evening will try again with 2025.3.0 and update you / provide build files.

syssi, 2025-03-21:
Short warning: The build folder will include all your secrets (WiFi password etc.) in plaintext. Please don't share the archive in public or make sure it doesn't include important secrets.

vit5041, 2025-03-21:
> Short warning: The build folder will include all your secrets (WiFi password etc.) in plaintext. Please don't share the archive in public or make sure it doesn't include important secrets.

I did the exersize again and confirm that compiling the same sketch with ESPHome 2025.3.0 adds webserver component into main.cpp.. See below, left side is ESPHome2025.2.2 and right side is ESPHome 2025.3.0
(do not provide full build directory due to secrets, but I have both, for old and new versions)

[image]

syssi, 2025-03-21:
I've seen your bug report. Just for reference: https://github.com/esphome/issues/issues/6883

vit5041, 2025-03-21:
> I've seen your bug report. Just for reference: [esphome/issues#6883](https://github.com/esphome/issues/issues/6883)

right, I opened another one, the issue can be reproduced with min sketch, but for lamp I have exactly this one:

https://github.com/syssi/esphome-yeelight-ceiling-light/blob/main/yeelight_light_ceil26.yaml

syssi, 2025-03-22:
I'm unable to reproduce the issue. This YAML doesn't add the `web_server` component to the build:

```
esphome:
  name: test

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

api:
```

What's different to your setup? How does the `wif` and `ota` section in your case?

vit5041, 2025-03-22:
> I'm unable to reproduce the issue. This YAML doesn't add the `web_server` component to the build:
> 
> ```
> esphome:
>   name: test
> 
> esp32:
>   board: wemos_d1_mini32
>   framework:
>     type: esp-idf

issue is with the board: esp32doit-devkit-v1

syssi, 2025-03-22:
I cannot reproduce the issue using

```
esphome:
  name: test

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  platform: esphome

logger:
  level: DEBUG

api:
````

too.

vit5041, 2025-06-14:
resolved in ESPHome